prusa3d/PrusaSlicer #11787: Accel to Decel for Klipper machines.

dobrzann, 2023-11-29:
I think it would be a good idea to add a simple function to handle the SET_VELOCITY_LIMIT ACCEL_TO_DECEL command with the ability to set a percentage value based on the M204 acceleration for devices using KLIPPER software, similar to how it is implemented in Orca Slicer. It would be beneficial to utilize the software without the need for additional scripts.

zhangrui1114, 2024-02-02:
Agree! Is there any scripts to add these percentage?

dobrzann, 2024-02-02:
> Agree! Is there any scripts to add these percentage?

I wrote a simple script that searches for the line 'M204 Sxxx' in the generated code and adds 'SET_VELOCITY_LIMIT ACCEL_TO_DECEL=xxx/2' below it, dividing it by 2. You just need to add the script's file path in the post-processing window. Certainly, this script should work in any slicer that supports post-processing and utilizes the 'M204 S' command.
[accel_to_decel_for_postprocessing.zip](https://github.com/prusa3d/PrusaSlicer/files/14136762/accel_to_decel_for_postprocessing.zip)

dobrzann, 2024-02-02:
Everything works well, regardless of whether you save the file to disk or send it directly to the printer

closed 2026-09-29 by prusaslicer-bot[bot]