prusa3d/PrusaSlicer #11788: Tooltips no longer available

reify3d, 2023-11-29:
### Description of the bug

Since upgrading to PrusaSlicer 2.7.0+MacOS-x64, hovering the cursor does nothing.
Expected behavior: tooltips should appear when hoving over slicer paramters
What actually happened: nothing

Version:   2.7.0+MacOS-x64
Build:     PrusaSlicer-2.7.0+MacOS-x64-202311231501

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 14.1.1 (Build 23B81)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   4.1.0
Profile:      Core
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 570X OpenGL Engine
GLSL version: 4.10.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_texture_mirror_once
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_mirror_clamp
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>

### Project file & How to reproduce

1. Launch PrusaSlicer 2.7
2. Hover a slicer parameter

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0+MacOS-x64

### Operating system

14.1.1 (23B81)

### Printer model

Prusa XL 5.1.0

dt63126, 2023-12-17:
FYI only, confirming the issue with the following system:

Prusa: 2.7.0+MacOS-arm64
Mac: Sonoma 14.1.2

lukasmatena, 2024-02-20:
Do you guys still have this issue? According to our testing it appears to have fixed itself with Sonoma 14.3 update. Can you confirm that?

neophyl, 2024-02-20:
Lukas I'm not on a MAC but can I direct your attention to this post on the Prusa Forum.  https://forum.prusa3d.com/forum/prusaslicer/what-does-warning-symbol-mean/
Tango still seems to be having the issue and that was 7 hours ago.

lukasmatena, 2024-02-20:
@neophyl Thanks, I asked him about what macOS version he is running. The fact that it is 7 hours ago does not mean that he already updated.

neophyl, 2024-02-20:
I realise that :)  He opened a different thread to start with asking what the warning triangle means ( https://forum.prusa3d.com/forum/prusaslicer/warning-sign-on-imported-object-to-slicee-what-is-the-warning/ ) before he found that older one about the tooltips.  Its why I directed him towards this github issue in that one where I pointed out that supplying basic info such as version on Mac OS might be useful information

TangoOversway, 2024-02-20:
No, I haven't updated - haven't been shown a notification for any updates from 14.2.1 at this time. I have checked and 14.3.1 is available. I'll update today or later tonight, when I have downtime. Also note that I'm not getting any messages about what PrusaSlicer is autorepairing in objects I load. (Yes, Tango there, TangoOversway here.)

reify3d, 2024-02-21:
> Do you guys still have this issue? According to our testing it appears to have fixed itself with Sonoma 14.3 update. Can you confirm that?

Yes it seems to be fixed. PS Version 2.7.2-alpha2+MacOS-x64, macOS Sonoma 14.3.1 (23D60)

closed 2024-02-21 by reify3d

TangoOversway, 2024-02-21:
Updated MacOS to 14.31 and PrusaSlicer is at 2.7.1+MacOS-arm64 and it's still there.

Was it fixed in any version before 2.7.2-alpha2?

TangoOversway, 2024-02-21:
I found a glitch with this. I am now at MacOS 14.3.1 and PrusaSlicer 2.7.2-Beta1.

If I load an STL through the menu, I get a tool tip telling me what PrusaSlicer did to fix the STL.

BUT - if I select the file in Finder and drag and drop it into PrusaSlicer, the tool tip does NOT appear. I still get the warning icon, but no tool tip with repair info. This is still frustrating since it's so much easier to drag and drop, especially when working with multiple STLs for one print that it would be a big help if it worked with drag and drop as well.

TangoOversway, 2024-02-21:
Sorry for the repeated posts on this, but this morning, after I've had time to work on this, I'm finding it inconsistent. Sometimes I get the tool tip that gives me the information and sometimes I don't. Even loading it through the menu, which is what I thought, at first, made it work (as opposed to drag-and-drop), but it seems that was just chance. I managed to get the tool tip a few times within the first 5-6 times I loaded an item with errors, but not consistently. At this point I haven't been able to get a tool tip for the last 6-8 times I've loaded the same STL object with errors in it.