prusa3d/PrusaSlicer #11790: Win Server 2019 2.7 does not run

rchadgray, 2023-11-29:
### Description of the bug

I use PrusaSlicer on a Windows Server 2019 and i RDP into the server to run the software.  My previous version 2.6 runs fine.  I upgraded to 2.7 and it installs, but will not run.  Are there logs i can check to see if it is crashing?

Installs and runs fine on my Win10 Laptop

I reverted back to 2.6.1 on my Win Server 2019 so i can get some work done.  

### Project file & How to reproduce

Install 2.7
It askes to un-install older version.  Yes
It installs with drivers
It never starts up.

I look at process list and do not see it running.  I have rebooted.

It just does not run.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7

### Operating system

Windows Server 2019

### Printer model

MK4

SaschaKP, 2023-11-30:
The problem is mainly into the PrusaSlicer.dll file, it won't get loaded, I bet there are some dependencies missing into a server machine, I myself have tried it, and it explicitly says that it won't load that .dll

rchadgray, 2023-11-30:
Did they do a complete re-write?  Looks like a new installer.  The version before works fine.

SaschaKP, 2023-11-30:
maybe this is related to #11781?

SaschaKP, 2023-12-01:
> I dont see any errors in Event viewer, where does PrusaSlicer log this?

just use prusa-slicer-console from a console window

sshockley, 2023-12-17:
Having the same issue on Server 2019 with 2.7.1, running prusa-slicer-console prints "PrusaSlicer.dll was not loaded"

sshockley, 2023-12-17:
Okay, through the magic of https://github.com/lucasg/Dependencies I was able to determine that PrusaSlicer 2.7 is looking for wlanapi.dll.  This is a new dependency added since 2.6.1.

On server 2019, you can add this in Powershell by running
```Powershell
Install-WindowsFeature -Name Wireless-Networking
```

You don't need to restart for PrusaSlicer to start working, and you don't need to enable the service.

Regardless, this is a bug in PrusaSlicer.  The program shouldn't exit silently when an optional Windows component isn't present.  I'm somewhat curious why a slicer needs to read the available wireless network list or read the wireless network profiles, but that's beyond the scope of this report.

SaschaKP, 2023-12-17:
> Regardless, this is a bug in PrusaSlicer. The program shouldn't exit silently when an optional Windows component isn't present. I'm somewhat curious why a slicer needs to read the available wireless network list or read the wireless network profiles, but that's beyond the scope of this report.

this is probably because it can send the file directly to the printer in XL and MK4, but anyway you're right, if it's not present, don't enable the feature and go on

lukasmatena, 2024-01-18:
Thanks, we will look into it.
SPE-2118

FidelCapo, 2024-01-29:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-01-29 by FidelCapo