prusa3d/PrusaSlicer #11792: Color Change in PS doesn't work correctly for Prusa XL

mkoistinen, 2023-11-30:
### Description of the bug

I'd like to do a 6-color print, so, at a layer after the last use of color 4 (initial brown), I added a color change in the slicer to change the color for tool 4. The current implementation in Prusa Slicer looks great! The pop-up asked me which tool and which color (blue), then when the part is displayed on-screen, it shows now all 6 colors. Brown for tool 4 in places below (lower Z) than the color change, and blue for layers above. Nice!

The printer did pause for the color change at the requested layer. Unfortunately, it seems I was only able to perform a color change on the then-selected tool, which was tool 5, not tool 4. I could not find any place in the "Tune" menu to change the filament for an arbitrary tool.

I'm using the recently release FW for the XL: 5.1.0

Might be related to #11600 

### Project file & How to reproduce

The file is larger than 25 MBs, so I put it into a public DropBox: https://www.dropbox.com/scl/fi/j8u0fi3xd9y0fm77wstm4/spongebob-PrusaXL.3mf.zip?rlkey=dbr0z1onz8rikgixcmqt377b0&dl=0

Here's me adding the color change for Tool 4:
[image]

Here it is with the color change, notice the iris of the eyes is tool 4, but is now blue.
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

macOS 14.1.1

### Printer model

Prusa XL

mkoistinen, 2023-11-30:
Also, I attempted to "fix" this by altering the color change Gcode in the advanced printer settings. I modified it to use:

```
P0 S1 L2 D0 ; Put away the current tool
"T3" + " S1 L0 D0
";  Select Tool 4
M600
P0 S1 L2 D0 ; Put away the current tool
"T" + current_extruder + " S1 L0 D0
"; Return to the original extruder
```

Rather than just M600. The Prusa XL seems to have entirely ignored this.

I then changed my color change marker in the slicer to a "Custom Gcode marker" with the above gcode and it was again ignored. Perhaps I have a typo in my gcode?

waldo1001, 2023-12-29:
This!

Jan-Soustruznik, 2024-01-04:
Hello, @mkoistinen, 

thank you for your report. We will check the problem. I closed this issue as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/11516

closed 2024-01-04 by Jan-Soustruznik