prusa3d/PrusaSlicer #11794: v2.7.0: prusa-slicer fails to link

nmschulte, 2023-11-30:
### Description of the bug

PrusaSlicer fails to link `prusa-slicer` binary on `version_2.7.0`:
```
[ 82%] Linking CXX executable prusa-slicer
/usr/bin/ld: /home/nmschulte/src/PrusaSlicer/deps/build-version_2.7.0/destdir/usr/local/lib/libtiff.a(tif_lerc.c.o): in function `LERCPostEncode':
tif_lerc.c:(.text+0xb49): undefined reference to `lerc_encodeForVersion'
/usr/bin/ld: /home/nmschulte/src/PrusaSlicer/deps/build-version_2.7.0/destdir/usr/local/lib/libtiff.a(tif_lerc.c.o): in function `LERCPreDecode':
tif_lerc.c:(.text+0x1331): undefined reference to `lerc_getBlobInfo'
/usr/bin/ld: tif_lerc.c:(.text+0x1431): undefined reference to `lerc_decode'
/usr/bin/ld: tif_lerc.c:(.text+0x1a97): undefined reference to `lerc_decode'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/PrusaSlicer.dir/build.make:176: src/prusa-slicer] Error 1
make[1]: *** [CMakeFiles/Makefile2:673: src/CMakeFiles/PrusaSlicer.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

It seems the TIFF update has caused this: 793e9f08fc704cbbcb2a4cf49b49ec090f27f921
Building on Debian GNU/Linux Trixie[/Sid] (13).

### Project file & How to reproduce

Build PrusaSlicer: https://github.com/prusa3d/PrusaSlicer/blob/version_2.7.0/doc/How%20to%20build%20-%20Linux%20et%20al.md

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

version_2.7.0

### Operating system

Debian GNU/Linux Trixie/Sid

### Printer model

Original Prusa MK4

nmschulte, 2023-11-30:
```diff
diff --git a/deps/+TIFF/TIFF.cmake b/deps/+TIFF/TIFF.cmake
index 9281dd5a6..88e0bbadc 100644
--- a/deps/+TIFF/TIFF.cmake
+++ b/deps/+TIFF/TIFF.cmake
@@ -10,6 +10,7 @@ add_cmake_project(TIFF
         -Dzstd:BOOL=OFF
         -Dpixarlog:BOOL=OFF
         -Dlibdeflate:BOOL=OFF
+        -Dlerc:BOOL=OFF
 )
 
 set(DEP_TIFF_DEPENDS ZLIB PNG JPEG OpenGL)
 ```

smurfix, 2024-02-16:
This fixes the WX problem:
```diff
diff --git src/CMakeLists.txt src/CMakeLists.txt
index 24534e75d..eb09007e8 100644
--- src/CMakeLists.txt
+++ src/CMakeLists.txt
@@ -47,9 +47,6 @@ if (SLIC3R_GUI)
         endif()
     endif()

-    if (CMAKE_SYSTEM_NAME STREQUAL "Linux")
-        set (wxWidgets_CONFIG_OPTIONS "--toolkit=gtk${SLIC3R_GTK}")
-    endif ()
     find_package(wxWidgets 3.2 MODULE REQUIRED COMPONENTS base core adv html gl)

     include(${wxWidgets_USE_FILE})
```

closed 2026-09-29 by prusaslicer-bot[bot]