prusa3d/PrusaSlicer #11800: Circle Travel Path with Extruded Material

Fortex365, 2023-11-30:
### Description of the bug

In the picture there are visible circles of travel path where the filament is extruded where it should not be. It all started from going from latest 2.6.1 to 2.7 release versions.

[image]

### Project file & How to reproduce

[racecar_final_simple_wheels.zip](https://github.com/prusa3d/PrusaSlicer/files/13514454/racecar_final_simple_wheels.zip)
In the attached file there is 3mf file with the model + printer settings. Go ahead and slice it to see.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7 release

### Operating system

Win 11 (23H2)

### Printer model

Ender3V2

neophyl, 2023-11-30:
Its the new helical ramp.  If you read the release notes it is mentioned.  So technically the fact it does so is not a bug.  

However the new feature has a few problems, for example https://github.com/prusa3d/PrusaSlicer/issues/11761 or https://github.com/prusa3d/PrusaSlicer/issues/11709 or even https://github.com/prusa3d/PrusaSlicer/issues/11754 although that one is a little different as its out of bounds but still, all 3 issues relating to the new feature.

johngalt01, 2023-12-19:
anyone know how to turn these  new helical ramps off!!!
i had to revert to 2.6 to get a project to complete without a major failure due to this new "bug" not feature.

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena