prusa3d/PrusaSlicer #11815: Extra perimeters on overhangs cause replacing bridges with concentric pattern

Le0Michine, 2023-12-01:
### Description of the bug

When I enable "extra perimeters on overhangs" for some model it causes slicer to treat bridging areas as overhangs rather than bridges which causes weird concentric pattern and poor print quality.

Below is the example of how bridging layer looks like with and without "extra perimeters on overhangs" (the difference is only that checkbox)

---
"extra perimeters on overhangs" enabled
[image]

---
"extra perimeters on overhangs" disabled
[image]

### Project file & How to reproduce

3mf file attached

try slicing it with and without "extra perimeters on overhangs".

[extra_perimeters_on_overhangs.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13530712/extra_perimeters_on_overhangs.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0+MacOS-x64

### Operating system

macOS 14.1.1 (23B81) Sanoma

### Printer model

N/A

mbrady4597, 2025-01-12:
Bump. I had this same issue causing lots of problems with large flat supported areas and just realized it was due to the "extra perimeters on overhangs" setting. I would recommend having a note to say experimental or describing the feature as in a beta test until the many issues with this setting are corrected.

Braintoe, 2025-04-03:
Still valid for the current version (2.9.1), Prusaslicer seems to forget to add bridges if this option is enabled:

[image]

Works fine though if disabled:

[image]

neophyl, 2025-04-03:
Well it IS marked as an experimental feature. 

[image]

Still.  That status hasn't changed.

Braintoe, 2025-04-03:
Yes, indeed. The intention behind the post was bump this up again and add some pictures since it took me a while to figure this out - I simply did not expect this feature to affect bridges of all things. 

The "Experimental" notice is good, but if the resulting issues are known (since years, in this case) it might be beneficial to clearly state somewhere (not necessarily inside Prusaslicer the online help for example might be fine as well) what you have to expect when activating it. At the moment, the help page the config option leads to when clicked on does not say anything about this feature

Braintoe, 2026-09-06:
Still interested in that (see last comment) therefore adding this comment on order to avoid closing the issue.