prusa3d/PrusaSlicer #11826: Feature Request: "Skip this version" when new version is available

RickSeiden, 2023-12-03:
**Is your feature request related to a problem? Please describe.**
When a beta or release candidate version is released, I get reminded there's a new version every time I start PrusaSlicer.  I normally wait until the official release, so getting this for beta and release candidates is annoying.

**Describe the solution you'd like**
I'd like an option to skip the release.

**Describe how it would work**
When you get the message that a new release is available, an additional link would be present allowing you to skip that release.  Once you select that, you won't be reminded about a new release until the next new release.

**Describe alternatives you've considered**
I've tried looking for an option in the settings, but all I can find is to turn it off altogether.  I want to be notified of new releases, but I don't want to be reminded every time I start PrusaSlicer that there is a beta or release candidate.

**Additional context**
This option would still give me a message every time a new beta or release candidate came out, but if I clicked the link to skip it, it wouldn't remind me until the next beta or release candidate came out.

Thank you!

Artemonim, 2023-12-03:
I think there needs to be an option to select which versions the user wants to be notified about:
- alpha+
- beta+
- rc+
- release only

lukasmatena, 2023-12-05:
The option mentioned by @JohnOCFII was implemented specifically for your use case, setting it to 'Release' will only notify you about the final releases. Can we close the issue?