prusa3d/PrusaSlicer #11836: Thumbnail generator should generate multicolor thumbnails

MisterGlass, 2023-12-03:
Between layer swaps and other features, it's becoming easier and easier to configure multicolor prints with PrusaSlicer, Like so:
[image]

Despite recent efforts in [#4577](https://github.com/prusa3d/PrusaSlicer/issues/4577) to allow changing the preview color, it still only supports generating single color previews, like so:
[image]

It would be great if thumbnail previews would include all selected colors, instead of a single primary color.

MisterGlass, 2023-12-12:
> It's also quite useful so you know which colors you will need

And what order to use them in as well.

Tupson444, 2026-09-18:
still wanted