prusa3d/PrusaSlicer #11837: "Collapsing overhang" is undocumented

timur-tabi, 2023-12-03:
### Description of the bug

There is no documentation anywhere that describes what a "collapsing overhang" is.  I asked this question on the forum:

https://forum.prusa3d.com/forum/prusaslicer/what-is-a-collapsing-overhang/

and I received no answers, so apparently no one else knows either.

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7

### Operating system

Linux

### Printer model

MK3s

sarusani, 2023-12-05:
I'm assuming it's an overhang that ends in "in the air" and would collapse when you print it without supports.

timur-tabi, 2023-12-05:
Maybe, but it would be nice to get an official answer and public documentation.  The truth is that the model prints just fine without supports, so I can't figure it out just by looking at it.

timur-tabi, 2026-09-03:
This still needs to be fixed.