prusa3d/PrusaSlicer #11841: supports not created correctly

greggjaskiewicz, 2023-12-04:
### Description of the bug

trying to setup organic supports for an item that's angled at 45/45 degrees - but the bottom of the cube is not having supports! I have to paint them in manually

### Project file & How to reproduce

[t_cube-t_cube_50mm.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13544266/t_cube-t_cube_50mm.3mf.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0+MacOS-arm64

### Operating system

macOS 14.1.2 (23B92)

### Printer model

mk4 running 5.1

robustini, 2023-12-05:
Even with SNUG he makes mistakes.
It puts a support inside the 2mm hole when it is not required by the supports I drew.

[image]

[Deriva DG600.zip](https://github.com/prusa3d/PrusaSlicer/files/13560242/Deriva.DG600.zip)

starthal, 2024-03-22:
The default "overhang threshold" is 40 degrees, so 45-degree overhangs won't be supported.

Arguably the only issue here is bed adhesion -- if the bottom corner were perfectly secured to the bed, the model would be printable as shown.

"Enforce support for the first N layers" should give you the necessary adhesion. Or using a raft. Or if the angle is the real issue, you can change the overhang threshold.

The slicer gives a warning about poor adhesion, but (AFAICT) only if support is off. Once you turn on supports it assumes you know what you're doing.

https://help.prusa3d.com/article/support-material_1698#overhang-threshold

closed 2026-09-29 by prusaslicer-bot[bot]