prusa3d/PrusaSlicer #11843: Prusa Slicer 2.70 wrong Lift Height value in GCODE after slice

yaronze, 2023-12-04:
### Description of the bug

Using Prusa Slicer 2.70 i noticed a problem with Travel Height after draw intro line to start printing position.
I took a look in the GCODE and compared it to previous version GCODE and i found out that in the "Before layer change" remark the PREVIOUS version value was -
G1 E-.8 F2100
G1 Z1.2 F10800
which was according to my settings in the project.
In 2.70 version the GCODE looks like-
G1 E-.8 F2100
G1 Z.2 F10800
though the Travel Height settings is any other value.
The value of Retraction Length is changing ok.
I can see for sure the difference between the slicer versions GCODE file.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.70 official version

### Operating system

 Windows 10 64 bit

### Printer model

Prusa MK3S

yaronze, 2023-12-05:
Hi
Thank you for your response. i will try to make it more clear. The new GCODE is -
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2
G1 E-.8 F2100
G1 Z.2 F720
;AFTER_LAYER_CHANGE
;0.2
G1 X110.139 Y91.083 F10800
G1 Z.2 F720
G1 E.8 F2100
M204 S800
;TYPE:Skirt/Brim
;WIDTH:0.42
G1 F1200
G1 X111.205 Y90.069 E.04613
G1 X112.5 Y89.746 E.04185

the height is always 0.2 while printing head is running to print position. This is not same as it was in previous versions and it causing problems in some of my challenging projects. In previous version i could lift to 1.2 mm in the first travel.

SachCZ, 2023-12-05:
Oh I see what you mean now. I have reproduced the issue. We will look into it.

yaronze, 2023-12-31:
Hi
Is there any option to change this value in the Prusa Slicer by the user? Or it is hard coded?
Since version 2.7.0 i have to modify gcode file manually to set the value otherwise some of my projects is failing and some time the printing sheet is seems to be damaged. 
Thanks

nmaggioni, 2024-01-09:
I am looking into this as well: it seems that v2.7.1 ignores the z-lift altogether, at least for the first layer:

Initial gcode produced by v2.7.1:

```gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
;0.2
G1 Z.2 F18000
;AFTER_LAYER_CHANGE
;0.2
_LAYER CURRENT=1
; printing object Shape-Box id:0 copy 0
G1 E-.75 F3000
G1 X164.193 Y164.193 F18000
G1 Z.2
G1 E.75 F1800
M204 S1000
;TYPE:Perimeter
```

Downgrading to 2.6.1 (temporarily removing Prusa's vendor file to skip the config incompatibility dialog) and reslicing the same test shape with the same profile gives:

```gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
;0.2
G1 Z.2 F18000
;AFTER_LAYER_CHANGE
;0.2
_LAYER CURRENT=1
; printing object Shape-Box id:0 copy 0
G1 E-.75 F3000
G1 Z10.2 F18000
M204 S1500
G1 X164.193 Y164.193
G1 Z.2
G1 E.75 F1800
M204 S1000
;TYPE:Perimeter
```

A `diff` makes the issue obvious:

```diff
@@ -13,7 +13,9 @@
 _LAYER CURRENT=1
 ; printing object Shape-Box id:0 copy 0
 G1 E-.75 F3000
-G1 X164.193 Y164.193 F18000
+G1 Z10.2 F18000
+M204 S1500
+G1 X164.193 Y164.193
 G1 Z.2
 G1 E.75 F1800
 M204 S1000
```

yaronze, 2024-01-12:
The problem is here - where after purge line the height is set to 0.2 and head travels to print start point very low. It was higher in previous versions.
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

G1 E-.8 F2100
**G1 Z.2 F10800**
;AFTER_LAYER_CHANGE

Is there any parameter or a way to modify this value in the slicer?

SachCZ, 2024-01-16:
Hi @yaronze  it is a bug in version 2.7.0 and there the value is indeed hardcoded. To fix the issue, the value will no longer be hardcoded in a future version, which means that any value from "start gcode" will be respected. It is currently planned for 2.7.2.

FidelCapo, 2024-01-29:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-01-29 by FidelCapo