prusa3d/PrusaSlicer #11845: PrusaSlicer 2.7 crashes when slicing big models

Maywtf12, 2023-12-05:
### Description of the bug

Since updating the PrusaSlicer to 2.7 I'm having problems slicing big models. I press "Slice Now", it goes to 20% maybe and then crashes without any message. It works just fine if I scale the model down substantially. Is there a way I can enable additional logging? I attached the event from windows event viewer.

Also full spec:
https://pastebin.com/bQXTByYB

### Project file & How to reproduce

[RG351P_Epic_ultra_Release_candidate_SD1.zip](https://github.com/prusa3d/PrusaSlicer/files/13560285/RG351P_Epic_ultra_Release_candidate_SD1.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.0+win64-202311231457

### Operating system

Windows 11 (build 22631), 64-bit edition

### Printer model

Prusa MK4

SckyzO, 2023-12-16:
Hello, Same issue here since prusaslicer 2.7.1 update. 
When click on "slice now",  it goes to 20% approximately and then crashes without any message.
For example, I use Phil Astronaut, size to 200% and slice (https://www.printables.com/model/634475-astronaut-phil-a-ment). 
Instant Crash when click on "Slice New". 
I uninstall Prusaslicer, remove all profile, reinstall, use a new default profile (artillery for example, i haven't mk3/4 :P), reslice et recrash ... 
Before, on release 2.6, PrusaSlicer never crashed ...

Absolo13, 2023-12-18:
I have the same issue

lukasmatena, 2023-12-18:
@Maywtf12 I was not able to reproduce it, it slices fine on my setup.

@SckyzO @Absolo13 Could you please provide a 3MF? Thanks.

KenneyDe, 2023-12-20:
Having same issue, When I try to slice a complex model it crashes on 20%.

lukasmatena, 2023-12-20:
Guys, we are thankful for your reports, but we need at least minimal cooperation in order to be able to find and fix them. Please, enclose your 3MFs, please. Please. We want to fix it!

SckyzO, 2023-12-20:
In my case the crashes happen with all the models I download. This is not related to a particular model, it is for all models, and even as soon as I had already printed with prusaslicer 2.6. The crash is relatively instantaneous, I click on Slice Now, and PS closes by itself, without an error message.

As I couldn't find where the error was coming from, I uninstalled PS and all my profiles (although I use that of Artillery X1 and FLSUN V400), and since then it seems to stabilize. I've had 1 or 2 other crashes, but it's no longer instantaneous. I wonder if it is not linked to profiles...

There should be an option in PrusaSlicer, when you install the program, it is to "empty PrusaSlicer of all profiles to start with a clean installation"

I'm going to do some tests again, select old profiles, and STLs and I'm going to add that to the ticket. What is certain is that I am not the only one and that this happens to quite a few people since version 2.7. Maybe it's related to a hardware conf too? I don't know.

KenneyDe, 2023-12-20:
[COLLAPSING-KATANA.zip](https://github.com/prusa3d/PrusaSlicer/files/13732235/COLLAPSING-KATANA.zip)
 Github does not support .3MF file type so I added it to .zip. I can replicate issue with other files that are more complex too not just this one.

SckyzO, 2023-12-20:
I confirm, i crash too when I slice this model :)

Maywtf12, 2023-12-21:
@lukasmatena it probably not caused by specific models, for me every bigger model crashes the Slicer the same. If I scale it down it slices just fine.

Is there a way for me to enable a debug mode or access additional logs?

KenneyDe, 2023-12-21:
[prusa-slicer.exe.44728.dmp](https://github.com/prusa3d/PrusaSlicer/files/13743027/prusa-slicer.exe.44728.dmp)
[prusa-slicer.exe.49124.dmp](https://github.com/prusa3d/PrusaSlicer/files/13743029/prusa-slicer.exe.49124.dmp)
Let me know if this works.

SckyzO, 2023-12-21:
[prusa-slicer.exe.53988.dmp](https://github.com/prusa3d/PrusaSlicer/files/13743448/prusa-slicer.exe.53988.dmp)
[prusa-slicer.exe.31252.dmp](https://github.com/prusa3d/PrusaSlicer/files/13743449/prusa-slicer.exe.31252.dmp)
[prusa-slicer.exe.35508.dmp](https://github.com/prusa3d/PrusaSlicer/files/13743450/prusa-slicer.exe.35508.dmp)

SckyzO, 2023-12-21:
STL + Dump file

[prusa-slicer.exe.52008.zip](https://github.com/prusa3d/PrusaSlicer/files/13743696/prusa-slicer.exe.52008.zip)

lukasmatena, 2023-12-21:
Thanks. The backtraces from @KenneyDe and @SckyzO are not the same.

@KenneyDe This is from your dump:
[image]
I assume that the problem will disappear if you disable Arachne perimeter generator.

@SckyzO Your dump:
[image]
This is a crash somewhere deep inside Clipper library, but I assume that the problem is really the data that Clipper gets from PrusaSlicer code. The minidump contains too little information to tell more.

lukasmatena, 2023-12-21:
It might also be worth looking if you are not running out of memory (by watching system monitor during slicing). But that should still be reported more gracefully than this.

KenneyDe, 2023-12-21:
> Hello @lukasmatena
> 
> Thank you for your feedback. My computer is an Intel I9 14900K, 32 GB DDR5 6800. So I think that's not the problem :) Of course, I tried swiping after reboot, with no apps open. Same problem.
> 
> How can I help provide you with more data?

I’m also running 14900k, 64gb ddr5 on my end, rtx4090.

KenneyDe, 2023-12-22:
Just an update, I have tried disabling Arachne perimeter generator (switch to classic) and still get the same crash at 20%. Also, I noticed if i zoom the camera to where the 3d object is no longer visible, it will progress past 20% and crash on 90% generating Gcode. Here are both dumps. (dump203706 is the 90% dump).
[prusa-slicer.exe.7260.dmp](https://github.com/prusa3d/PrusaSlicer/files/13747303/prusa-slicer.exe.7260.dmp)
[prusa-slicer.exe.20376.dmp](https://github.com/prusa3d/PrusaSlicer/files/13747304/prusa-slicer.exe.20376.dmp)

lukasmatena, 2023-12-22:
Thanks for all the data. Is it ok in  2.6.2? Does it work in the 2.7.0-alpha? Can you guys also test which is the first PrusaSlicer version where it crashes?

lukasmatena, 2023-12-22:
Could you please try out the following build? It is very wild guess and it will probably not solve the problem, but it is worth trying. [PrusaSlicer-2.7.1+3-lm-mysterious-clipper-crashes-win64-g945677b92](https://slicerbuilds.prusa3d.com/PrusaSlicer-2.7.1+3-lm-mysterious-clipper-crashes-win64-g945677b92-202312221129.zip)

Thanks.

Maywtf12, 2023-12-22:
> Could you please try out the following build? It is very wild guess and it will probably not solve the problem, but it is worth trying. [...]

<Error>
<Code>NoSuchKey</Code>
<Message>The specified key does not exist.</Message>
<Details>No such object: prusa3d-content-prod-14e8-slicerbuilds/PrusaSlicer-2.7.1+3-lm-mysterious-clipper-crashes-win64-g945677b92-202312221129.zip</Details>
</Error>

Maywtf12, 2023-12-22:
> > Is there a way for me to enable a debug mode or access additional logs?
> 
> You are on Windows, meaning you can enable local crash dump collection. The crash dump would show us where the application crashes. If you want to spend time with it, it would be very helpful. See [the respective wiki page](https://github.com/prusa3d/PrusaSlicer/wiki/Enabling-crashdump-collection-on-Windows) for instructions.
> 
> > KenneyDe
> 
> Thanks, but I am not able to reproduce it with your 3MF either.

@lukasmatena 
[prusa-slicer.exe.46112.dmp](https://github.com/prusa3d/PrusaSlicer/files/13753797/prusa-slicer.exe.46112.dmp)

lukasmatena, 2023-12-22:
@Maywtf12 Sorry, my bad. I fixed the link in my previous post.

newlandk, 2023-12-30:
I'm running 2.7.1 on Windows 11 on a 13900k, 32gb DDR5, RTX4090 and experiencing crashes when slicing that sound like the same symptoms as OP with the app crashing shortly after the slicer starts. I am able to slice the exact same Prusa 3mf file on a Win10 virtual machine on a different server as well as an older laptop running Windows 10.

I got an inkling to try a CPU limit using my motherboard utility since I did notice an quick fan ramp on my CPU AIO before crashing. I went into my BIOS and the configuration for the CPU power was essentially set to unlimited and after specifying the Intel POR limits (~253 Watts) it hasn't crashed since. I've repeated the test 3 times with and without limits and it continues to crash when power is not limited (even though I never see the charts go above 255 watts, but that could be a sampling issue).

I did try running with the integrated graphics disabled earlier and it had no effect. I'm not sure what the Slicer could be doing differently than other intensive applications...  maybe using a different/new instruction set...

Either way... it's working crash-free now and I'm hoping this helps someone else's issue as well! Take care!

SckyzO, 2023-12-30:
Wow, that's funny, today I did the same thing. I used XTU to reset overclocking and add CPU limit, and after that no crashes. I actually think we are making progress on the bug.

Maywtf12, 2023-12-31:
> @Maywtf12 Sorry, my bad. I have fixed the link in my previous post.
[prusa-slicer.exe.41088.dmp](https://github.com/prusa3d/PrusaSlicer/files/13802356/prusa-slicer.exe.41088.dmp)

This version also crashes the same way, uploaded a fresh dump.

lukasmatena, 2024-01-02:
Also, I have two builds with two separate attempts to fix it. Could you please test them both and tell me whether it helps or not?
[2.7.1+3-lm-mysterious-clipper-crashes-win64-g945677b92](https://slicerbuilds.prusa3d.com/lm/PrusaSlicer-2.7.1+3-lm-mysterious-clipper-crashes-win64-g945677b92-202312221129.zip)
[2.7.1+1-lm-updated-tbb-win64-g1f6f352df](https://slicerbuilds.prusa3d.com/lm/PrusaSlicer-2.7.1+1-lm-updated-tbb-win64-g1f6f352df-202401021232.zip)
Thanks.

Stealey24, 2024-01-02:
I've been having the same exact issues and was able to fix it. Like @newlandk said above I think the issue is related to using "high-end rigs". This is my new work computer and I don't know much about it but the IT guy who built it was able to fix the issue. He said went into BIOS and "turned off sync all core overclock and switched to 'auto'." I hope this helps. If you need more info I'll try my best to find it. 
Edition	Windows 10 Pro
Version	22H2
OS build	19045.3803
Processor	13th Gen Intel(R) Core(TM) i9-13900K   3.00 GHz
System type	64-bit operating system, x64-based processor

natn, 2024-01-02:
> Also, I have two builds with two separate attempts to fix it. Could you please test them both and tell me whether it helps or not?  

2.7.1+3-lm-mysterious-clipper-crashes-win64-g945677b92 build -  I have had five consecutive successful slices when loading the sample .3mf project I uploaded in the other thread.  However, when I load the source stl into a blank project and enable organic supports it still crashes silently.  Both behaviors have been fully repeatable over about 5 attempts 

2.7.1+1-lm-updated-tbb-win64-g1f6f352df build -  one successful slice from the 3mf project, but then three consecutive silent crashes after that.

As far as I know, I am not having issues with any other applications.

natn, 2024-01-02:
> 2.7.1+3-lm-mysterious-clipper-crashes-win64-g945677b92 build - I have had five consecutive successful slices when loading the sample .3mf project I uploaded in the other thread. However, when I load [...]

Interestingly, If I add the source stl, enable organic supports, save a new .3mf project, close PS, reopen PS and load the saved file, the crash so far has not occured over 3 attempts and the slice was successful.

It's not conclusive, but the "mysterious-clipper" build seems to work with a project loaded from a .3mf, but not if I build the project manually and slice without a .3mf save and a reload.

lukasmatena, 2024-01-03:
After reading through

[reddit:i913900k_instability_crashes_and_svid_behavior](https://www.reddit.com/r/intel/comments/16vp5kn/i913900k_instability_crashes_and_svid_behavior/)
[community.intel.com: i9-13900K-very-frequent-crashes-Windows-11-with-apps-games](https://community.intel.com/t5/Processors/i9-13900K-very-frequent-crashes-Windows-11-with-apps-games-and/m-p/1528232/highlight/true#M65549)
[reddit: im_having_some_serious_issues_with_my_i9_14900k](https://www.reddit.com/r/overclocking/comments/18mc8xd/im_having_some_serious_issues_with_my_i9_14900k/)

and others, I am starting to think that it really could be a hardware issue. The common denominator in all these strange bug reports are the i9-13900K / i9-14900K processors. The crash dumps that some of you have posted lead to various places without an obvious connection - they all end somewhere in the Clipper library, but that can be a coincidence, given that Clipper is frequently used during the slicing process. This would indicate that they occur more or less randomly, when the CPU is under heavy load.

I am not even slightly a hardware expert, but reading though the online discussions would indicate that the default BIOS settings are set at the edge of where the hardware is stable, maybe in an attempt to make the high-end CPUs look like even higher-end. Who knows. These are the things that some people report as solutions in similar cases (BIOS settings):
- set SVID Behaviour to "Intel’s Fail Safe"
- set Turbo Power Limits to "Intel POR"
- limit P cores to 5.7 GHz

**Could you all please try that and report whether it helps? And I am still looking forward to when someone will share the information when the problems started. If it is ok in e.g. 2.6.1, it would indicate that the bug is really in PrusaSlicer, or that we at least started doing something that provokes it. Thanks.**

> It's not conclusive, but the "mysterious-clipper" build seems to work with a project loaded from a .3mf, but not if I build the project manually and slice without a .3mf save and a reload.

This is quite confusing. Some defects in the model may have been repaired during the save-load, so it is possible that the model from the 3MF is (slightly) different, but I don't see the connection with the changes in the "mysterious-clipper" build.

silence48, 2024-01-20:
the issue happens in prusa slicer, bambu studio, superslicer, so seems that whatever it is, likely is underlying from slic3r.

Delberan, 2024-01-26:
I also have started getting the same crashes. Also using an I9-14900K processor.
Started sometime after the 2.7.1 update.

SckyzO, 2024-02-06:
Good morning,

the best way I found to stabilize PrusaSlicer is to tune the CPU a little thanks to XTU (then if you want to put it in the bios to save the parameters hard, that's no problem).
I didn't touch the CPU power, but I did force a power limit and add a little voltage to the CPU cores (for better stability).
For me it works. Be careful before touching anything you must have a minimum knowledge of computers, BIOS, CPUs :D

[image]

Nerdcaught, 2024-02-12:
Hey everyone! so Im not actively using Prusaslicer, I use Orcaslicer but I am having the same issues as described here. I hopped over to Prusaslicer to see if the same crash occured and copied as many settings as I could over from Orcaslicer. In a file that the slicer states has more then 1M triangles even when simplified it crashes.  https://www.thingiverse.com/thing:6382941 I have to cut the mask in half in order to fit my build plate. 

I recently upgraded from a I7-10700k to a I7 13700k both with 128gb of RAM. Both systems would crash but only when it had any type of Overclock, I mess around with some BIOS settings when overclocking, but if I just leave AI Overclocking with the 13700k the programs do not crash.  I believe its definitely some sort of Hardware issue but its a spike that occurs when the walls are being generated.  It feels like that activates a spike of some kind that the programs do not like. My 10700k used to BSOD the PC, while my 13700k now just crashes both prusa and orca.  Im going to try the suggestion above from [lukasmatena] and report back on prusaslicer.

Nerdcaught, 2024-02-13:
I tried the suggestion and it did not work out for me. I did get BSOD every time i tried a different SVID behavior option. Unless I left on Auto which I found odd. I think the spike might not be on the CPU side, but on the RAM side. if I leave it on XMP profile base it crashes . but recently I starting messing around with basic overclocking of my RAM. Im running 64gb Trident Z RGB My XMP profile is 3600 mhz at 1.35v  it crashes. but if I set it at 1.45v-1.5v and increase it to 4000mhz or 4266mhz in the BIOS, it does not crash and slices quite quickly.  idk if anyone else wants to try that out.

jcleek, 2024-02-20:
I opened an issue for the crashing as I was unaware of this thread. Thank you @lukasmatena for your recommendations to test. I only had to make the one change to SVID behavior.

Motherboard: ASUS ROG STRIX Z790E GAM WF D5
Processor: INTEL INTEL I9-13900K 
RAM: 32GB G.SKILL Trident Z5 RGB Series (Intel XMP 3.0) DDR5 RAM 32GB (2x16GB) 6400MT/s CL32-39-39-102 1.40V Desktop Computer Memory UDIMM
Video Card: ASUS ROG Strix NVIDIA GeForce RTX™ 4070 Ti OC Edition Gaming Graphics Card

In the BIOS I changed: EXPERT -> AI TWEAKER -> SVID BEHAVIOR -> INTEL'S FAIL SAFE

PrusaSlicer now works as expected.

newlandk, 2024-03-01:
I had a similar experience @jcleek. My solution that is still working is to set the core voltage to a fixed value manually and lowered the Turbo Boost maximum power draw. In my case I chose 1.425 volts and 200 watts.

I have a 300mm AIO and don't see exorbitant temperatures... so I wouldn't expect whatever is happening to be thermal related. I'm wondering if something is hiccuping with power delivery when the load hits the CPU. 

What's odd is the regularity of being able to trigger this with a slicer... I have had one or two instances where I've seen similar behavior when under other loads, but Prusaslicer seems to trigger without fail when slicing a large file. Are there particular instructions that are being used that would behave differently than a standard benchmark or stressing utilities?

jcleek, 2024-03-02:
Replaced my processor today with a i9-14900K and immediately tried to slice the same model I have been having issues with this week. Sliced without issue. I put in more and more complexity in the slicing and it didn't fail or crash one time. Guess it is time to reach out to Intel to get a replacement or refund.

laurentslab, 2024-03-03:
> In the BIOS I changed: EXPERT -> AI TWEAKER -> SVID BEHAVIOR -> INTEL'S FAIL SAFE
> 
> PrusaSlicer now works as expected.

13900K on an Asus Z790 ProArt. Same issue and this suggestion fixes the problem. I picked SVID > Worst Case, which also works on my config.

ke19lly, 2024-03-11:
I was able to fix this by limiting the Program to my 14900K cores 0-7. My steps are as follows: open task manager, go to the details pane, scroll down to find Prusaslicer, right-click Prusaslicer and select set affinity, select only cores 0-7 or the respective amount for lower-end LGA 1700 chips, then confirm the selection and Prusaslicer works as expected.

ke19lly, 2024-03-11:
> @ke19lly <https://github.com/ke19lly> that only works per session. If I
> close Prusa Slicer and reopen I have to do that again. Any chance you found
> a more permanent "fix"?

No, I haven't found a more permanent solution yet. I've submitted a feature
request asking for a compatibility option to be added, but for now, it
seems we're stuck with this temporary fix.

jcleek, 2024-03-11:
I have a more permanent "fix".

1) Locate your prusa-slicer.exe file. This is the executable and is likely located here: C:\Program Files\Prusa3D\PrusaSlicer
2) Right-click it and select CREATE CHORTCUT. If you don't see CREATE SHORTCUT you are on WIN11 and you should first click SHOW MORE OPTIONS -> CREATE SHORTCUT
3) You will probably see an error that you can't create a shortcut there and ask to create one on the desktop. Answer YES.
4) Find the shortcut on your desktop. Right-click it and select PROPERTIES.
5) In the TARGET box add this in front of what is there:
          %ComSpec% /C Start "PrusaSlicer" /Normal /Affinity FF
   So the TARGET box should look something like this:
          %ComSpec% /C Start "PrusaSlicer" /Normal /Affinity FF "C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe"
6) Click CHANGE ICON
7) In the top box LOOK FOR ICONS IN THIS FILE put in the path to your prusa-slicer.exe file like this: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe and press ENTER
8) You should see the icon for PrusaSlicer. Make sure it is selected and click OK
9) Click OK
10) Now double-click the shortcut. Check the affinity by looking at the details in Task Manager as @ke19lly indicated above.

This isn't a fix, just a good workaround for PrusaSlicer.

ke19lly, 2024-03-11:
I attempted that workaround yesterday but encountered an issue where the
shortcut wouldn't launch the slicer. I'll give it another try, as it would
indeed be convenient to avoid manually setting the affinity each time.
Thank you for the detailed instructions.

Taralas, 2024-03-19:
> I opened an issue for the crashing as I was unaware of this thread. Thank you @lukasmatena for your recommendations to test. I only had to make the one change to SVID behavior.
> 
> Motherboard: ASUS ROG STRIX Z790E GAM WF D5 Processor: INTEL INTEL I9-13900K RAM: 32GB G.SKILL Trident Z5 RGB Series (Intel XMP 3.0) DDR5 RAM 32GB (2x16GB) 6400MT/s CL32-39-39-102 1.40V Desktop Computer Memory UDIMM Video Card: ASUS ROG Strix NVIDIA GeForce RTX™ 4070 Ti OC Edition Gaming Graphics Card
> 
> OS: Windows 11 Pro Version 22H2, OS build 22621.3155, Windows Feature Experience Pack 1000.22684.1000.0
> 
> PrusaSlicer: Version 2.7.1+win64
> 
> In the BIOS I changed: EXPERT -> AI TWEAKER -> SVID BEHAVIOR -> INTEL'S FAIL SAFE
> 
> PrusaSlicer now works as expected.

Same result here. with a slightly different motherboard, memory, and GPU:

Motherboard: ASUS ROG MAXIMUS Z790 FORMULA
Processor: INTEL INTEL I9-14900K
RAM: 64GB G.SKILL Trident Z5 RGB Series (Intel XMP 3.0) DDR5 RAM 32GB (2x16GB) 6400MT/s CL30-40-40-96 1.40V Desktop Computer Memory UDIMM
Video Card: ASUS TUF GeForce RTX™ 3090

arnfiix, 2024-04-28:
> I have a more permanent "fix".
> 
> 1. Locate your prusa-slicer.exe file. This is the executable and is likely located here: C:\Program Files\Prusa3D\PrusaSlicer
> 2. Right-click it and select CREATE SHORTCUT. If you don't see CREATE SHORTCUT you are on WIN11 and you should first click SHOW MORE OPTIONS -> CREATE SHORTCUT
> 3. You will probably see an error that you can't create a shortcut there and ask to create one on the desktop. Answer YES.
> 4. Find the shortcut on your desktop. Right-click it and select PROPERTIES.
> 5. In the TARGET box add this in front of what is there:
>    %ComSpec% /C Start "PrusaSlicer" /Normal /Affinity FF
>    So the TARGET box should look something like this:
>    %ComSpec% /C Start "PrusaSlicer" /Normal /Affinity FF "C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe"
> 6. Click CHANGE ICON
> 7. In the top box LOOK FOR ICONS IN THIS FILE put in the path to your prusa-slicer.exe file like this: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe and press ENTER
> 8. You should see the icon for PrusaSlicer. Make sure it is selected and click OK
> 9. Click OK
> 10. Now double-click the shortcut. Check the affinity by looking at the details in Task Manager as @ke19lly indicated above.
> 
> This isn't a fix, just a good workaround for PrusaSlicer.

This fixed (or at least bypassed) the issue for me, thanks!

MackBambu, 2024-05-28:
bambulab/BambuStudio#3318

Maywtf12, 2024-05-28:
TL;DR Intel cranked up its CPUs to appear better on benchmarks than AMD, making them unstable.

Hello,
@lukasmatena  thank you for the investigation and the tips.
This issue is indeed associated with 13/14th gen Intel i9 processors. 
AMD has recently made some pretty good CPUs that are now cheaper, more power-efficient and with better performance at times. I believe it led Intel to some unhealthy practices of cranking up their CPUs to unhealthy values, just to appear better on benchmarks. That's why you will only find people having issues with their i9s, as this is the competing product. I had a hard time finding someone with the same problem on i7.

These issues will cause your CPU to crash with various errors, the most common being "failed to allocate video memory". This led a lot of people to create tickets for their video card manufacturers. Nvidia did their own investigation and concluded that the Intel CPUs are at fault. You can see their advice in [driver patch notes](https://www.nvidia.com/en-us/geforce/forums/game-ready-drivers/13/540532/geforce-grd-55212-feedback-thread-released-4424/).

The solution is to set the appropriate power limits in BIOS. Refer to your motherboard manufacturer for guidance - all of them released guides and baseline profiles to help you with that. Prolonged, unmitigated usage of these CPUs can lead to more serious issues like silicon degradation. In my case, I has to RMA my CPU, Intel sent me a new one. If you set the power limits and you still have this problem - create a ticket to Intel. They will most likely advice you to offset the voltage on cores, but if your system is not stable on default values, you are entitled for a replacement.

We are still waiting for a public statement from Intel, because as of now they only tried to put the blame on someone else.

More info:
[Reddit post](https://www.reddit.com/r/overclocking/comments/1axepvu/optimizing_stability_for_intel_13900k_and_14900k/)
Some youtube hardware journalists:
[JayzTwoCents](https://www.youtube.com/watch?v=HIubZYwBfPc)
[Hardware Unboxed](https://www.youtube.com/watch?v=OdF5erDRO-c)
[The Provoked Prawn](https://www.youtube.com/watch?v=Nm6fXM5Qk60)

SckyzO, 2024-05-28:
Just read this news : 
- https://www.theverge.com/2024/4/9/24125036/intel-game-crash-13900k-14900k-fortnite-unreal-engine-investigation
- https://www.theverge.com/2024/5/8/24152383/intel-motherboard-manufacturer-i9-stability-bios-update-baseline-profile-default
- https://www.techradar.com/computing/cpu/intels-core-i9-cpus-have-been-having-some-serious-issues-but-intel-insists-its-your-motherboards-fault
- https://www.tomshardware.com/pc-components/cpus/intel-reportedly-demands-all-board-partners-implement-intel-default-baseline-profile-by-may-31-company-hopes-to-fix-issues-with-some-core-i9-chips
- https://cdrdv2.intel.com/v1/dl/getContent/743844?explicitVersion=true

In my opinion, and i did this and it's work for me, I just update with new bios on my Z790 Gigabytes Aorus Elite AX (https://www.gigabyte.com/fr/Motherboard/Z790-AORUS-ELITE-AX-rev-10/support#support-dl-bios). Others motherboards have new bios too, but READ THE MANUAL before update.

ayeohdee, 2024-06-28:
I have this same problem with a high end setup. I disable e-cores via task manager / set affinity for PS and the crashing stops. With the 13900k I only leave 0-7 enabled and I do not crash at 20% or at any percent anymore.

jcleek, 2024-06-28:
> I have this same problem with a high end setup. I disable e-cores via task manager / set affinity for PS and the crashing stops. With the 13900k I only leave 0-7 enabled and I do not crash at 20% or at any percent anymore.

Look up for my response on March 11th for a more permanent "fix" you can use.

chuckkay, 2024-07-01:
> * %ComSpec% /C Start "PrusaSlicer" /Normal /Affinity FF "C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe"

thanks that works, but of course only when opening slicer with that shortcut, not when opening slicer with 3mf file association (double click on 3mf file" i will try to make a regedit for associated filetypes

RoccoFL, 2024-07-15:
Another 13900K user for whome the Affinity Trick fixed my slic3r based issues.

jcleek, 2024-07-22:
Intel finally figured it out: [https://www.xda-developers.com/intel-cracks-the-cause-raptor-lake-instability/](url)

R3iZoR, 2024-08-20:
I am running on an i9-14900KS and I was having headaches with the slicer crashing at 80% of the times!

I just want to say that the latest intel 0x129 microcode patch fixed all the stability issues for me, no longer needed to run the prusa slicer on specific affinity or eco mode.

Update your Bios guys!

closed 2026-09-29 by prusaslicer-bot[bot]