prusa3d/PrusaSlicer #11848: "Avoid Crossing Perimeters" fails in some cases (maybe on a short distance)

TheGoofy, 2023-12-05:
### Description of the bug

I've turned on the feature to "avoid crossing perimeters", but in some cases it's having no effect
[image]
There is certainly a better (and even shorter) path, which would even avoid the perimeter.

### Project file & How to reproduce

[RW Ratchet Wheel.zip](https://github.com/prusa3d/PrusaSlicer/files/13572527/RW.Ratchet.Wheel.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0 win64

### Operating system

Windows 11

### Printer model

Ultimaker 2

petaflot, 2024-06-09:
I have a quite similar issue, where the travel moves are completely absurd (unnecessary zigzags, going back and forth a few times) and sometimes it does a very good job (even using curved paths?).

A clue for a fix would be to use existing paths on the same (or next) layer

[image]

see also #8441

JanHouwers, 2024-06-26:
The setting: Avoid crossing perimeter setting has issues.

When printing with LW-PLA this issue is really annoying.

[image]

Projectfile and repreduction
[test4.zip](https://github.com/user-attachments/files/15989099/test4.zip)

Prusaslicer 2.7.4

Windows 10

Printermodel: Prusa MK4 and XL

jmd73, 2024-06-26:
@JanHouwers The problem seems to be the parameters "Filaments->Filament Overrides->Retraction->Minimum travel after retraction" and "Printers->Extruder 1->Retraction->Minimum travel after retraction" that are set to 30 in your profiles.

By unchecking it in the overrides and setting it to default "1.5" by pressing the back arrow, it looks better.
From filament settings:
[image]

From printer settings:
[image]

And the result:
[image]

JanHouwers, 2024-06-26:
Thanks for the pointer... In my case it improves these prints a lot.

closed 2026-09-29 by prusaslicer-bot[bot]