prusa3d/PrusaSlicer #11859: file locked

feanorgem, 2023-12-06:
### Description of the bug

When I open a file from File Explorer by clicking on it, the file gets locked so that I can not do a save and must save it as a new file. Even after Prusa slicker closes, the file remains locked and I have to erase it from the command prompt. It also happens when I start slicer and open a recent file.

### Project file & How to reproduce

ANY file will do

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.70. and the previous 2.6.x

### Operating system

win10

### Printer model

pruce 3si

lukasmatena, 2023-12-06:
PrusaSlicer cannot keep any lock on a file after it is closed. If a file is opened, it is opened by a process that is currently running. I am not sure if I understand the issue description completely, but you mention something about a File Explorer, which reminds me of similar issues caused by the file preview function, as described e.g. [here](https://answers.microsoft.com/en-us/windows/forum/all/cant-delete-file-because-of-preview-pane-in/f7d18d6d-0120-44c8-a391-b720086aa72f). Could that be the problem?

lukasmatena, 2023-12-06:
> perhaps its because they don't open the file directly but make a temp copy that they work out of?

3MF files are not opened as a copy. Again, if there is some problem when PrusaSlicer is closed, I would look elsewhere. Especially given that the 2.6.x is out for a long time and no similar issue was reported so far, at least as far as I know.

>  Notepad++, office, PDF, etc they all have no problem with file preview

Those applications do not open 3MF files. Don't you use some 3MF preview plugin that would do this? Could it be related to antivirus settings?

feanorgem, 2023-12-06:
what I'm saying is that the applications that don't have a problem (notepad++) DO allow previewing with out locking problems so they are doing OPEN different, either as a share or a copy or something but they way slicer is doing it is WRONG. I don't know how Windows does preview of files but I'm betting they use slicer somehow

feanorgem, 2023-12-06:
I'm also saying something has changed. This didn't use to happen. I haven't updated my Win10 since before this started, only slicer has updated.

feanorgem, 2023-12-06:
The preview shows the same render as when opening slicer and sence it worked in 2.5 erra, slicer has changed something, that is broke, bug

 
 

If you are talking about the preview of 3MF files in Windows File Explorer, PrusaSlicer has nothing to do with it. But maybe I am misunderstanding the description.

CD1010, 2023-12-10:
I have the same issue.  
[image]
[image]
I have prusa slicer 2.7 on windows 10.

closed 2026-09-29 by prusaslicer-bot[bot]