prusa3d/PrusaSlicer #11860: Unable to build PrusaSlicer for Windows. MSB8066 error custom build CMake wxWidgets

DanielPuchta, 2023-12-06:
### Description of the bug

I cloned the Prusa git exactly as it is described - https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md

Then I followed these instruction (instructions 2A - Manual Build Instructions):

cd c:\src\PrusaSlicer\deps
mkdir build
cd build
cmake ..
cmake --build .

In the last line I encountered problem. The MSVC x64 Native Tools Command Prompt compiled and build the project, but it ended with the error -  MSB8066 error custom build CMake wxWidgets. The error message:

" D:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(247,5): e
rror MSB8066: Custom build for C:\src\PrusaSlicer\deps\build\CMakeFiles\58cc1f77be6cd0aadc493af67cd02f17\dep_wxWid
gets-mkdir.rule;C:\src\PrusaSlicer\deps\build\CMakeFiles\58cc1f77be6cd0aadc493af67cd02f17\dep_wxWidgets-download.rule;C
:\src\PrusaSlicer\deps\build\CMakeFiles\58cc1f77be6cd0aadc493af67cd02f17\dep_wxWidgets-update.rule;C:\src\PrusaSlicer\d
eps\build\CMakeFiles\58cc1f77be6cd0aadc493af67cd02f17\dep_wxWidgets-patch.rule;C:\src\PrusaSlicer\deps\build\CMakeFiles
\58cc1f77be6cd0aadc493af67cd02f17\dep_wxWidgets-configure.rule;C:\src\PrusaSlicer\deps\build\CMakeFiles\58cc1f77be6cd0a
adc493af67cd02f17\dep_wxWidgets-build.rule;C:\src\PrusaSlicer\deps\build\CMakeFiles\58cc1f77be6cd0aadc493af67cd02f17\de
p_wxWidgets-install.rule;C:\src\PrusaSlicer\deps\build\CMakeFiles\61bed3dbbaf0357936b26fa68277567c\dep_wxWidgets-comple
te.rule;C:\src\PrusaSlicer\deps\build\CMakeFiles\9b14d2d62fba49f2f00cf042a3e0829e\dep_wxWidgets.rule;C:\src\PrusaSlicer
\deps\CMakeLists.txt ended with code 1. [C:\src\PrusaSlicer\deps\build\dep_wxWidgets.vcxproj]"

I downloaded wxWidgets library and instaled it (using "Build" command in VisualStudio). Error still persist. How to install wxWidgets library? In order to successfully build PrusaSlicer to be able to add new functions and to modify existing code? What am I doing wrong?

### Project file & How to reproduce

Clone Prusa git on new computer with clean windows 10 and follow exactly the build instructions in readme.md or here: https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md

That means no preintalled libraries, just VisualStudio 2022 with C++ desktop, and Cmake.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 10

### Printer model

Prusa i3mk3S

DanielPuchta, 2023-12-06:
Do you mean logs inside CMakeFiles folder? Full path: "C:\src\PrusaSlicer\deps\build\builds\CURL\CMakeFiles"? 

Let me know if those are the files you meant.

[CMakeError.log](https://github.com/prusa3d/PrusaSlicer/files/13589501/CMakeError.log)
[CMakeOutput.log](https://github.com/prusa3d/PrusaSlicer/files/13589502/CMakeOutput.log)

lukasmatena, 2023-12-06:
Yes. It is not wxWidgets which fails to build, it is CURL, namely its tests. And it looks suspicious, because the errors (there are many) seem to indicate that it compiles code that it should not compile on Windows at all.

I would start with reinstall MSVC, but honestly, I have no idea if that helps. Either way, it looks like something on your end and I cannot promise that we will have time to look into it more closely.

lukasmatena, 2023-12-08:
To be honest, I don't know what the problem is. The step by step guide you followed should be enough, I am not aware of any pitfalls related to MSVC, CMake and git installation.

RaymondKroon, 2023-12-10:
I think I have the same problem. I think it is because of the newly released cmake version 3.28, which has different CMP0057 default behaviour. Installing 3.27.9 solves it. But the real fix should be in the config.

```
cmake --build .
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Performing configure step for 'dep_wxWidgets'
  CMake Deprecation Warning at CMakeLists.txt:10 (cmake_minimum_required):
    Compatibility with CMake < 3.5 will be removed from a future version of
    CMake.

    Update the VERSION argument <min> value or use a ...<max> suffix to tell
    CMake that the project does not need compatibility with older versions.

  -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.22631.
  -- EXPAT found in C:/Development/PrusaSlicer-deps/usr/local/lib/cmake/expat-2.4.3
  CMake Warning (dev) at C:/Users/raymo/scoop/apps/cmake/3.28.0/share/cmake-3.28/Modules/FindTIFF.cmake:118 (if):
  -- Configuring incomplete, errors occurred!
    Policy CMP0057 is not set: Support new IN_LIST if() operator.  Run "cmake
    --help-policy CMP0057" for policy details.  Use the cmake_policy command to
    set the policy and suppress this warning.

    IN_LIST will be interpreted as an operator when the policy is set to NEW.
    Since the policy is not set the OLD behavior will be used.
  Call Stack (most recent call first):
    build/cmake/functions.cmake:608 (find_package)
    build/cmake/options.cmake:116 (wx_add_thirdparty_library)
    build/cmake/main.cmake:16 (include)
    CMakeLists.txt:100 (include)
  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at C:/Users/raymo/scoop/apps/cmake/3.28.0/share/cmake-3.28/Modules/FindTIFF.cmake:118 (if):
    if given arguments:

      "CXX" "IN_LIST" "TIFF_FIND_COMPONENTS"

    Unknown arguments specified
  Call Stack (most recent call first):
    build/cmake/functions.cmake:608 (find_package)
    build/cmake/options.cmake:116 (wx_add_thirdparty_library)
    build/cmake/main.cmake:16 (include)
    CMakeLists.txt:100 (include)
 ```

DanielPuchta, 2023-12-11:
> I think I have the same problem. I think it is because of the newly released cmake version 3.28, which has different CMP0057 default behaviour. Installing 3.27.9 solves it. But the real fix should [...]

You are right, it worked with cmake version 3.27.9 I hope that this issue would be helpfull :)

closed 2023-12-11 by DanielPuchta

xkuyax, 2024-02-29:
@DanielPuchta how did you install cmake 3.27.9 on windows with visual studio?

pmarks-net, 2024-03-03:
> how did you install cmake 3.27.9 on windows with visual studio?

Brute force solution: run `where cmake` and replace the files in that directory.

"You are right ... Cmake version 3.28 is not working" is not a good reason to close a bug as Completed.

pmarks-net, 2024-03-03:
I had better luck building with Visual Studio 2019.