prusa3d/PrusaSlicer #11862: not possible to click in 'repair' icon inside prusaslicer 2.7.0 in linux

otaolafranc, 2023-12-07:
### Description of the bug

If I try clicking in the warning icon of the model (as it can be seen in the image) instead of repairing the model as it commonly does in windows, in linux it prompts me to change the name of the model. this is an issue as it is only way to acces to the repair feature....
[image]

### Project file & How to reproduce

adding a model with an issue in its mesh, and trying to repare it.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.0+linux-x64-GTK3-202311231454

### Operating system

linux Tuxedo OS 2

### Printer model

mk4

sarusani, 2023-12-07:
The repair feature is using a windows service for fixing the errors. It's not available on linux.

rtyr, 2023-12-07:
As sarusani mentioned, this option is available only in Windows.

closed 2023-12-07 by rtyr