prusa3d/PrusaSlicer #11869: A Weird Bridging pattern

Ivan-753, 2023-12-08:
### Description of the bug

bridges are sliced in a strange way...
[image]

### Project file & How to reproduce

standard built-in benchy,
0.4mm nozzle,
0.2mm layer height
PrusaSlicer-2.7.0+linux-x64-GTK3-202311231454.AppImage

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Linux

### Printer model

Tronxy

neophyl, 2023-12-08:
The reason you need to attach a project when opening an issue is that it will contain ALL the settings you are using.  For example if I slice a 'standard' benchy with a 'normal' profile that area is sliced normally like so
[image]

With your project the devs can tell what is going on hopefully.  

From experience though if I had to GUESS , the cause is the EXPERIMENTAL setting 'extra perimeters on overhangs'.  Turn that OFF.

Ivan-753, 2023-12-08:
here it is:

BUGED if:
Detect bridging perimeters: on
perimeter_extrusion_width: 0,45

OK if:
Detect bridging perimeters: off
perimeter_extrusion_width: 0,45

OK if:
Detect bridging perimeters: on
perimeter_extrusion_width: 0,4
[Detect bridging perimeters bug.3mf.zip]

(https://github.com/prusa3d/PrusaSlicer/files/13615473/Detect.bridging.perimeters.bug.3mf.zip)

closed 2026-09-29 by prusaslicer-bot[bot]