prusa3d/PrusaSlicer #11870: Independent linear/pressure advance for internal and external perimeters.

MartijnGevaert, 2023-12-08:
**Is your feature request related to a problem? Please describe.**
Bulging on seams, especially for thick layers.

Bulging on seams is reduced by tuning linear/pressure advance. However, these values are tuned for a specific speed, acceleration, ... Specifically, it's recommended to tune these for the external perimeter settings. This means that the linear advance values are not tuned for the internal perimeters.
Most of the time this is not an issue, but on thick layer lines (0.4mm layer w. 0.6 nozzle), the bulging of the internal perimeters pushes out the seam of the external perimeter.
The more internal wall perimeters, the more obvious this issue should become.

**Describe the solution you'd like**
I believe if the internal perimeters could have a different PA value, tuning could be done separately for internal and external perimeters, having more filament being put down with it's optimal PA value.

**Describe how it would work**
Be as descriptive as your skill level allows you.
The slicer would simply have to emit a gcode value whenever it switches between internal/external perimeters.

**Describe alternatives you've considered**
There are other solutions regarding flow around the seems, such as stopping earlier, coasting, ... although they work in practice, they are more of a patch on underlying differences between the modeled flow and actual flow.

**Additional context**
/ Let me know if there is any additional investigation/information which I can add which would be of use here.
Unfortunately I can't test the separate PA values in practice yet because it's not trivial to add these gcode commands manually as far as I know.

MartijnGevaert, 2023-12-13:
A workaround was provided over at the SuperSlicer, but it's also usable on PrusaSlicer:
https://github.com/supermerill/SuperSlicer

Ataraxia-Mechanica, 2024-06-11:
https://github.com/SoftFever/OrcaSlicer/pull/5609
This is being developed for orca slicer, and it do make a difference in print quality.

Wegerich, 2026-09-18:
I still think this should be implemented in prusaslicer