prusa3d/PrusaSlicer #11871: Speed on wipe tower attached to the filament material

palimbro, 2023-12-08:
**Is your feature request related to a problem? Please describe.**
Yes, i need to use petg as a support for a PP object, the problem is that the petg adheres poorly to the PP if the speed is too high, but I don't want to slow everything down because for the rest of the printing, everything goes well.
The wipe tower comes off because the layers have not properly adhered to each other, if I reduce the overall speed, it works but my entire print is much slower.
It will be very usefull, like PLA+TPU, PETG+TPU ....... 
I have MMU2S and XL (5tools)
Exemple here with PETG+PP:
[image]

**Describe the solution you'd like**
adding an adjustment to the speed of the wipe tower by material would be most suitable 

**Describe how it would work**
just adding an option on all materials filaments

**Describe alternatives you've considered**
don't found an alternative

andyylin, 2024-07-01:
Wipe tower speed was set to perimeter and infill speed in 2.6.1, but like @palimbro I also am having issues with PETG sticking to PLA on my wipe tower. I also want to be able to set the speed of the wipe tower separately.

andyylin, 2024-07-05:
Here's my lovely PETG and PLA wipe tower. [image][image]

closed 2026-09-29 by prusaslicer-bot[bot]