prusa3d/PrusaSlicer #11872: [2.7.0] Supports generated where not required

robustini, 2023-12-09:
### Description of the bug

I have verified that in many cases supports are generated in places where it was not requested.
In the attached design you can check in a moment, I have the requirement that the center hole is not filled by the supports and yet it happens.
The problem is evident with any kind of support.
Even inserting the negation keeps ignoring it.

[image]

https://github.com/prusa3d/PrusaSlicer/assets/3864712/d4565a1e-3995-4ef9-ad3c-d30b3f0dc0f6

### Project file & How to reproduce

[Deriva DG600.zip](https://github.com/prusa3d/PrusaSlicer/files/13623250/Deriva.DG600.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

V2.7.0

### Operating system

Windows 10 Pro x64

### Printer model

Delta

neophyl, 2023-12-09:
I'm not a dev, but I took a look at your project as it was interesting.  With your chosen support type of organic and the limits on that support it really doesn't have any other choice on how to construct that tree.

All an enforcer and blocker does is define **surfaces** on the model to support and not support.  The support is then generated downwards from there.  Blockers are not a keep out area.  The painted blocker is being respected, its not touching the model in that area but to support the areas around that you have defined as needing support it will grow to where ever needed.

The tree on the inside is the same.  You haven't enabled support from build plater only so its allowed to start a tree on your model if it thinks it needs to to support an area you have defined and there's no wall sliced there so the tree can get there

Frankly I cant see why you are using organic for this job anyway, Snug would work much better and wouldn't go through the hole (especially if you turned the first layer expansion to zero and altered the closing distance on the support settings)

[image]
This is with Snug and a bit of settings adjustment.

I have attached a project for you to examine too 
[Deriva DG600_neo.zip](https://github.com/prusa3d/PrusaSlicer/files/13624257/Deriva.DG600_neo.zip)

robustini, 2023-12-10:
> Frankly I cant see why you are using organic for this job anyway

I use organic because it helps me to have anchor points on the bed to keep the print solid or it starts to wobble when it gets very high.

amatulic, 2024-01-16:
I have seen this problem with organic and snug supports but not grid supports.

Seems to be related to #11251 and #11818