prusa3d/PrusaSlicer #11873: Move binary gcode and label objects to printer settings

KayGestalten, 2023-12-09:
Hi,
as I didn't found a related ticket. I highly suggest to move the bgcode and the label object settings from print settings to the printer settings. Currently I had to change them everytime I switch the targeted printer, because until some days octoprint arn't able to use bgcode. Also the naming of the objectes are different between octoprint and the prusa firmeware.

The reason is pretty simple, theee print settings are mostly related to the physical printer. So depending on the firmare that they are running or if I uses octoprint or direct connection.

So it would be pretty easy if these settings where at the printer, so when I'm switch the printer these where changed too.

Kay

Tupson444, 2023-12-09:
It's mentioned that bgcode should be moved to printer settings in #11734, but otherwise it's not the same issue.

I agree that these two settings should be in the printer settings, it makes much more sense.

edwardhorsford, 2023-12-09:
Just came here to investigate this.

I've got a 0.4mm and 0.6mm printer profile, with and without input shaping. I think that means I would need to go through all 4, editing every print profile for each, and then ending up with *duplicates* just to disable this and use Octoprint. Should someone really need to edit ~24 profiles and have them show as duplicates to disable this feature? PrusaSlicer already knows they're being sent to Octoprint.

Tupson444, 2023-12-09:
@edwardhorsford Are you always using Octoprint, or just sometimes? If always, then you would need to set up the printer profiles only once. Correct me if I'm wrong (I haven't used Octoprint so I don't know).

In any case, having these in print settings is arguably worse, as someone may have only 2 printer profiles and 50 print profiles (which are not dependent on printer profile). In that case, the 50 print profiles would need to be duplicated. Alternatively, if these settings are in printer profiles, then they would only be set once without making duplicate profiles (each profile with appropriate settings for that printer / configuration / nozzle size).

edwardhorsford, 2023-12-09:
@Tupson444 I've updated my comment to be clearer that this change seems sensible.

KayGestalten, 2023-12-09:
@Tupson444 not only you, I took it in the same direction 😄

timur-tabi, 2023-12-09:
This is a duplicate of bug 11734.  We're all asking for the same thing -- move the setting to Printers, and make it off by default.

timur-tabi, 2023-12-09:
It would be better to change this bug to only talk about "label objects" and use bug #11734 for the bgcode option.  That way, each request can be evaluated separately.

edwardhorsford, 2023-12-09:
Neither seems to be a bug, and #11734 advocates a different solution - to disable BGCode by default. This issue advocates an _alternative_ to #11734. 

Even if #11734 were done, moving the setting for BGCode to printer settings would still make sense for people who want to use it.

closed 2023-12-14 by lukasmatena

KayGestalten, 2023-12-14:
@lukasmatena about the gcode, thats nice. But whats about the object labeling?

AudreyAP, 2024-01-27:
It does not make sense to have object labelling in print settings. +1 to moving object label to printer settings.