prusa3d/PrusaSlicer #11874: Could not connect to Duet: HTTP 404:

boelle, 2023-12-09:
### Description of the bug

when i click test under host upload i get: Could not connect to Duet: HTTP 404
i just updated to 2.7.0 and did not have the issue with 2.6.1

[image]
[image]

webinterface works fine

[image]

not sure what's going on

### Project file & How to reproduce

this happens without any stl loaded

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Win11

### Printer model

self build with re-arm controller and rpi running duet firmware

boelle, 2023-12-09:
i can of course save the gcode to desktop and upload it myself, but i liked the option to upload direct from the slicer

lukasmatena, 2023-12-11:
Thanks for the report. Could you please run PrusaSlicer from the console with enabled logging by running `prusa-slicer-console.exe --loglevel 5`, let the upload fail and post the log from the console window? It might give some clues about what the problem is. Thanks.

kocikdav, 2023-12-12:
Hello @boelle , thanks for the log. Perhaps this could be a bug with incorrect password. The documentation states that default password is "reprap", but lets try not including it. We have prepared a custom [build](https://storage.cloud.google.com/prusa3d-content-prod-14e8-slicerbuilds/PrusaSlicer-2.7.0%2B1-dk-duet-fix-win64-g466fdc8cf-202312120857.zip). Would you test it for us? 
Just unzip the file and run prusa-slicer-console.exe. Than if the problem remains, please copy the log again. Thank you.

boelle, 2023-12-12:
"Perhaps this could be a bug with incorrect password"

i never used password, so it must be something else

lukasmatena, 2023-12-12:
@boelle [This](https://slicerbuilds.prusa3d.com/PrusaSlicer-2.7.0%2B1-dk-duet-fix-win64-g466fdc8cf-202312120857.zip) should be the correct link.

boelle, 2023-12-12:
[image]

[image]

boelle, 2023-12-12:
Q: not a programmer, but has anything changed in regard to this from 2.6.1 to 2.7.0 ?

boelle, 2023-12-13:
testing to see what version breaks

2.6.1

[image]

[image]

kocikdav, 2023-12-13:
Thank you very much for the logs. Seem like changes for https://github.com/prusa3d/PrusaSlicer/issues/10061, specifically _GET /machine/connect_ does not work for you. We will fix it.

boelle, 2023-12-13:
my hardware setup is a re-arm controller with reprap firmware attached to a rpi 

Duet Web Control 3.3.0

Board: rearm (LPCSBC)
DSF Version: 3.3.0
Firmware: RepRapFirmware for LPC176x based Boards 3.3.0_11 (2021-11-04)

not sure if that is a common setup or not, but i used to have marlin firmware on it and they changed the config file so often that i gave up and looked for something else. https://teamgloomy.github.io/getting_started.html gave me the option to run my setup as a DUET and i have only heard positive things. Gave it a try and have never looked back since

as they write there should be no difference between my setup and a real duet, well maybe updates since version 3.3.0 could have made things different, but since the port for re-arm will not be updated i will remain at 3.3.0, works quite well and maybe someday i will have enough money that i can afford to trash working hardware and get a real duet

or maybe i'm lucky enough that i could become a tester for prusa and get one of their machines instead of the self build stuff i run on

Seicosid, 2023-12-18:
Hi @kocikdav,

I know you guys are looking into this, but I came across the exact same issue on my setup as @boelle, however, my hardware differs slightly.

Duet board conectged to rpi:
Board: Duet 3 MB6HC (MB6HC)
DSF Version: 3.3.0
Firmware: RepRapFirmware for Duet 3 MB6HC 3.3 (2021-06-15) 

Just an fyi really, but if there is an update we can test, please let me know

Thanks!

closed 2026-09-29 by prusaslicer-bot[bot]