prusa3d/PrusaSlicer #11875: Minimum bridge wall length

rozhkovets, 2023-12-09:
**Is your feature request related to a problem? Please describe.**
For example, when I print with a lot of infill, the distance between the infill lines is very small, sometimes less than 1 mm. But due to the set low speed of printing bridges, the top layer above such infill is also printed at low speed. Why slow down if, with such filling, the top layer will turn out good and will not sag?

**Describe the solution you'd like**
Cura use the "minimum bridge wall length" parameter. Unsupported walls shorter than this will be printed using the normal wall settings. Longer unsupported walls will be printed using the bridge wall settings.

For example, simple cube 20 mm with 99% infill. 
[image]
[image]
**Additional context**
Add any other context or paint drawing about the feature request here.

petrisi, 2024-11-12:
I would like to see this feature happening as well for fan speed reasons.

Tupson444, 2026-09-15:
still relevant