prusa3d/PrusaSlicer #11878: UI Drop down menu list length shorter in 2.7.0

Shmidtey, 2023-12-10:
### Description of the bug

In prusaslicer version 2.7.0 the dropdown menu/list length is shorter than it used to be in previous versions. Not sure if this would be a bug or a feature request, since it was already functioning as expected in 2.6.1. See included pictures.
I searched and could not find a similar issue, so apologies if I missed it.

In 2.6.1 the drop down menu/list for things like filaments or printers used to have enough room for 30 filaments/lines. 
After 2.7.0 there is only room for 15 filaments/lines.

I would consider this undesirable, because it requires more scrolling to find the filament I'm looking for.
I find it to be much less of an annoyance to just click the drop down, and at glance be able to see most of the filaments. Even if I don't see the filament I'm looking for, at least I get a better idea of where I am in the list. Is that making sense?

I would even prefer if I could make the list longer than 30 lines, as I have monitor space to spare. But I'll settle for having it back to 30 😅.
Changing application text size has no effect on the number of lines the dropdown lists display. 

I also double checked that my windows resolution and display scaling were not involved, changing them had no effect on the list length.
This is my first report, so I hope I included all the relevant details.

### Project file & How to reproduce

Have 2.7.0 installed, and more than 30 filaments configured.
Click filament dropdown list
[image]
[image]
[dropdown menu bug report project file.zip](https://github.com/prusa3d/PrusaSlicer/files/13626187/dropdown.menu.bug.report.project.file.zip) (Empty- no object)
[config.zip](https://github.com/prusa3d/PrusaSlicer/files/13626184/config.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.0+win64

### Operating system

Windows 10 Pro

### Printer model

Original Prusa Mini+

Tupson444, 2023-12-10:
I don't remember how it was before, but I think it's too short now, requiring too much scrolling. If screen size is not tiny, the list should be way longer as it is much more practical that way.

stillpointx, 2023-12-15:
This is also happening in Linux in prusaslicer 2.7.1  I can't even see what mine fully says anymore  

2.6.1
[image]

2.7.1
[image]

Stefan-Odw, 2023-12-15:
Yes, same for me. It would be nice to have a longer list visible.

closed 2026-09-29 by prusaslicer-bot[bot]