prusa3d/PrusaSlicer #11879: Park tool during heat absorbtion phase

Jeggo65, 2023-12-10:
**Is your feature request related to a problem? Please describe.**
Park tool when print bed is in heat absorbing phase to reduce/avoid oozing of filament.

**Describe the solution you'd like**
Please add the following code in the printer start G-Code
`P0 S1 L1 D0; park the tool to prevent oozing <-- New
G29 G ; absorb heat <-- Current
; select previous tool again  <-- New
T{initial_tool} S1 L0 D0 <-- New`

**Describe how it would work**
Extra G-Code has been successfully tested on a 5 head XL with FW 5.1.0 and Prusa Slicer 2.7.0.
Please include in future versions.

**Describe alternatives you've considered**
Move print head out of the way and seal nozzle when not needed. Same for G-Code M601. Tool should be parked

**Additional context**
N.A.

rtyr, 2023-12-11:
Heat absorbtion time is based on the filament type. Initial tool needs to be picked so FW knows what time to choose.

I am not sure whether this is actually needed, but if so, it should be handled by the firmware, not by start gcode. Same goes for M601 behavior.

closed 2023-12-11 by rtyr

Jeggo65, 2023-12-11:
This is not only a feature request this is also an issue for me. When I print with PETG, the string that oozes out of the nozzle during heat absorbtion leads to a failure of the nozzle cleaning feature ofter the heat absorbtion in nearly 50% of the time.
As the tool is picked before the heat absorbtion starts, the time is correct and not influenced by the parking of the tool.
If it is a firmware feature or slicer features is really not clear. If it is a slicer feature, you could easily adapted it to your needs.

Jeggo65, 2023-12-11:
Yes, about 10-20mm (Prusament PETG Orange). It is enough to confuse the nozzle cleaning after the heat absorbtion.