prusa3d/PrusaSlicer #11882: Add support for Elegoo SLA Open-Source Goo Format

JohannesSteiner, 2023-12-10:
**Is your feature request related to a problem? Please describe.**
Currently, PrusaSlicer supports various third-party SLA printers. Unfortunatly, recent Elegoo SLA printers (like the Elegoo Mars 4 DLP) are currently not covered since they exclusively understand the new "Goo Format".

**Describe the solution you'd like**
The official Goo Format Specification is available [here on Github and maintained by Elegoo](https://github.com/elegooofficial/GOO/tree/main). It would be awesome to have this integrated into PrusaSlicer or at least have a workarround to use PrusaSlicer with new Elegoo printers.

**Describe how it would work**
I guess same as for all third-party SLA formats?

**Describe alternatives you've considered**
UV-Tools already support Goo-Format. If there is a convenient way to automate slicing via UVTools, this would of course be sufficient.

Thank you very much in advance!
A frustrated Elegoo Mars user

Felix-Rm, 2024-02-23:
There is currently a PR (#12251) open to implement the requested features.

Archives of the built application with the new features can be found [here](https://github.com/Felix-Rm/PrusaSlicer/releases/tag/test-build.goo-format-support-dev). Alternatively you can of course also build from source.

Please check out the PR for advice on setting up the printer profiles. People with Elegoo Mars 4 printers are welcome to test and report their findings in the PR.

JohannesSteiner, 2024-02-23:
Thank you! This is awesome! I will give it a go within the next days!

closed 2026-09-29 by prusaslicer-bot[bot]