prusa3d/PrusaSlicer #11886: v2.7.0: Partial bridge infill lines being generated on a single internal layer.

sp847242, 2023-12-11:
### Description of the bug

v2.7.0: The part mostly sliced normally, but on one layer (layer 66, 9.95mm), some long bridge infill lines were generated inside of the part. PrusaSlicer says "No errors detected" in the STLs themselves. 

The .3MF file is attached. 

This is the only part I've seen this occur on.

If I change the infill style to Rectilinear, there are still some unusual bridge infill lines made on layer 66 toward one side of the part.

### Project file & How to reproduce

Reproducing the issue: All that should need to be done is to load the attached 3MF, click Slice, and check Layer 66.

[image]
[part.zip](https://github.com/prusa3d/PrusaSlicer/files/13638517/part.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 10

### Printer model

Prusa MK3S+

rtyr, 2023-12-12:
It looks technically correct (closest anchor point), even though it seems odd a little bit.

j28CLY, 2026-09-18:
This is still present in 3.0.0-alpha11. The results aren't identical, but they are similar.
The original .3mf file won't load for me though, "Loading file failed: Cannot find compatible printer HW config."

A 3.0.0-compatible part file is attached. The internal bridging is on layer 65 here.

[part, 3.0.0-alpha11.zip](https://github.com/user-attachments/files/32381318/part.3.0.0-alpha11.zip)