prusa3d/PrusaSlicer #11888: Create (folder) path via `Output filename format` and `Placeholders`

thomaslagies, 2023-12-11:
**Is your feature request related to a problem? Please describe.**

When uploading files via `PrusaLink` or even directly onto an USB stick, I like to put the sliced `gcode` files 
within their respective `filament/toolhead` folders.

Given the situation that you have multiple printers, or often change filament types, you have to change the 
folder name while exporting the sliced file.

e.g. from `/PLA/<filename>.gcode` to `/PETG/<filename>.gcode`

I tried to include `/` into the output filename, but it gets removed and the placeholders left of the `/` are ignored in the filename

**Describe the solution you'd like**

Treat `/` in `Output filename format ` as folders that shall be created either directly on disk, or via `PrusaLink` onto the `remote` USB stick

**Describe how it would work**

`/` shall create folders via Prusa Slicer when saving to disk, or when using the `Send to printer` button.

[image]

[image]

[image]

t-makaro, 2023-12-14:
This would be great. I always like to upload all of my gridfinity prints to the same "Gridfinity" folder via PrusaLink. But setting ```Gridfinity/{input_filename_base}_0.4n.gcode``` as the output filename format does not work, and I'm sick of changing it all the time. It would be really nice to tie the output directory to a specific print profile. This may require making the ```Remember Output Directory``` setting specific to print settings instead of global.

jothepro43, 2025-05-28:
This would be a nice feautre. I recently tried doing this with PrusaSlicer connected to Octoprint. Will this be implemented possibly in the near future?

thomaslagies, 2026-09-03:
so, is it already possible to sort output files in different folders. Based on parameters?

jothepro43, 2026-09-03:
I'm curious to know as well.