prusa3d/PrusaSlicer #11890: Arrange of Large Objects on the XL places item outside print area

BrienKing, 2023-12-12:
### Description of the bug

If I use the standard arrange:

[image]

Then it puts my second object outside the printable area when it would fit just fine on the bed.
[image]

I can manually arrange them just fine.
[image]

### Project file & How to reproduce

See the attached images above.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 10

### Printer model

Prusa XL Dual Extruder

sarusani, 2023-12-12:
You defined it should use a 6mm spacing between objects. That makes them not fit on the print bed. Nothing wrong with the arrange feature.

BrienKing, 2023-12-12:
Each Square on the picture is 10mm.  That gives about 60mm top to bottom.   Well within the 6mm spacing.

sarusani, 2023-12-12:
Are you using a skirt and or brim?

Please just provide the project file, so people don't have to guess your settings...

BrienKing, 2023-12-12:
[CaseBottom.zip](https://github.com/prusa3d/PrusaSlicer/files/13642944/CaseBottom.zip)

sarusani, 2023-12-12:
Thanks.
You're right, I can reproduce the issue also with unmodified standard profiles.

closed 2026-09-29 by prusaslicer-bot[bot]