prusa3d/PrusaSlicer #11891: Configurable exported G-code filename

rem821, 2023-12-12:
Hey, recently I've been exporting a lot of gcodes and been manually adding infill percentage into the filename. I thought it could be a good idea to give the user the ability to change the default filename pattern and possibly extend it with some information that the slicer knows during exporting (infill patern, infill percentage, supports) if the user chooses to.

lukasmatena, 2023-12-12:
Like this?
[image]

rem821, 2023-12-12:
Hey! Thanks for a quick response. Shame on me, I thought I've searched everywhere but I've somehow missed this tab.
This is exactly what I've meant.

closed 2023-12-12 by lukasmatena