prusa3d/PrusaSlicer #11894: Incorrect filling of hollow object

LibThi, 2023-12-13:
### Description of the bug

When slicing, the hollow object is filled. Previous versions do not.
[image]
[image]

### Project file & How to reproduce

[Krytka.zip](https://github.com/prusa3d/PrusaSlicer/files/13658106/Krytka.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Verze 2.7.0+win64

### Operating system

Windows 11, v. 23H2

### Printer model

Prusa i3 MK3S

neophyl, 2023-12-13:
No bug here.  The software is doing exactly what it is supposed to be doing.  Your settings are 
[image]

You have slicing Mode Close Holes selected.  Change it to regular and you get
[image]

LibThi, 2023-12-13:
It´s true 🫣

closed 2023-12-13 by Jan-Soustruznik