prusa3d/PrusaSlicer #11897: Please allow us to change the Extrusion Multiplier in the middle of the print and reset it to default of 1 after print

laptop-laptop, 2023-12-13:
**Is your feature request related to a problem? Please describe.**

I want to print with Extrusion Multiplier X% from first layer to a certain layer height. Then, change the Extrusion Multiplier to another percentage value until the end. Finally, reset it back to 100% at the end of the print.

I know I can set the value from Filament Settings -> Filament -> Extrusion multiplier. That stays throughout the entire print job. I want to change its value at certain layer height.

**Describe the solution you'd like**
Tried to do it via Height Range Modifier but there is no option to change the Extrusion Multiplier value.

**Describe how it would work**
Please see above

neophyl, 2023-12-13:
You can already do this.  

Slice your model, use the layer slider to select a layer (just like if you were going to do a colour change) and then right click the plus icon instead of left clicking the plus.  That will bring up the 3 options 
[image]

Choose Add custom Gcode and then enter in the flow rate command you want to go to.  For example for a marlin based printer you could use M221 S150 to set the flow rate to 150%.  
Add as many custom gcode commands as required at whatever layers you require.

To reset between prints just modify the printer Start gcode block field with a M221 S100 that will ensure that every new print starts with the flow at 100%.  Profiles for prusa printers like the MK3 already have a M221 in there to set the flow to a specific level at the start anyway.

rtyr, 2023-12-13:
Answered.

closed 2023-12-13 by rtyr

laptop-laptop, 2023-12-13:
Thank you. On the internet, some people said the flow rate (parameter for M221 and Flow on the LCD of Prusa i3 MK3S+) is not the same as Extrusion Multiplier and we need to calculate the value. What is the official answer for that?