prusa3d/PrusaSlicer #11899: Dovetail cutting works in one direction but not when object (and cut) rotated 90 degrees

puterboy, 2023-12-13:
### Description of the bug

I am able to cut a long object in Dovetail mode when the object is parallel to the longer edge of the plate but when I rotate the object (and the cut) by 90 degrees in the plane, it fails to allow the cut.

### Project file & How to reproduce

Note you can see in the pictures below that all the settings are the same other than the 90 degree rotation.
In the first image, "Perform Cut" is active while in the second image it is not.

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7

### Operating system

Windows 11

### Printer model

Prusa MK3S+

puterboy, 2023-12-13:
Here is a simplified project file - it just consists of an elongated box (i.e., beam) shape that I created in the slicer itself
Cut about halfway using:
Z=4.5
Depth = 2.25mm + .10mm
Width = 2.25mm + .10mm
Flip Angle = 60deg
Groove Angle = 0deg
[dovetailtest.zip](https://github.com/prusa3d/PrusaSlicer/files/13665995/dovetailtest.zip)

YuSanka, 2023-12-22:
Thanks for your report.
But there is intended behavior. You need to have a valid tongue and groove, when cut is in Dovetail mode.

puterboy, 2024-01-02:
> Thanks for your report. But there is intended behavior. You need to have a valid tongue and groove, when cut is in Dovetail mode.

Your reply doesn't make sense.
The cut is the same in both images -- i.e., both the object and the cut plane have been rotated 90 degrees in the plane of the print bed -- so if the here is a valid tongue-and-groove in the first picture then there *should be* a valid one in the second picture.
HOWEVER, there is not which is why I filed this bug report!!!!
There is no seeming reason why one should work and the other doesn't.

Please re-read my note and re-look at the clear pictures posted.

YuSanka, 2024-01-05:
@puterboy, I think, that It really make sense ;) Just look at dovetail closely.

First variant of cut plane
[image]
Zoom-in it and you can see valid cut
[image]

And second variant of cut plane
[image]
Zoom-in it and you can see NON-valid cut (some edges are outside of object)
[image]

puterboy, 2024-01-05:
I see that now... but there is still a problem or at least unexpected/unexplained behavior.
In the original rotation, the dovetail is along the "vertical" face of the object being cut while after rotating the object 90 degree (in the plane of the print bed), the dovetail is now on the "horizontal" face which corresponds to a 90 degree rotation orthogonal to the bed plane.

I would have thought that rotating the object in the bed plane would similarly rotate the orientation of the dovetail 90 degree in the bed plane.

What am I missing and how do I fix this?
More generally, how do you purposefully rotate the plane of the dovetail?

neophyl, 2024-01-06:
You use the tools/handles that are part of the cut tools.  When selected you can rotate in the axis selected
[image]

Like the normal rotate tool the compass lines can be snapped to so you can easily get 90/45 and 5 degree increments.

puterboy, 2024-01-07:
Ahhh, I didn't realize that -- it looked the same as the cut rotation so I thought it just changed the cut rotation. Thanks!

HOWEVER, IT STILL DOESN"T EXPLAIN WHY rotating the object 90 degrees in the X-Y (i.e. PRINT BED) PLANE rotates the dovetail 90 degrees in the (orthogonal) Z-Y plane -- this is highly counter-intuitive and unexpected.

puterboy, 2026-09-02:
Keep this alive until fixed!