prusa3d/PrusaSlicer #11901: 2.7.0 Supports are generated mid air

axandar, 2023-12-13:
### Description of the bug

When generating supports, slicer starts them in air, not on baseplate. Problem was partially fixed when brim was disabled - on one model it started creating support material from plate.
Layer 5:
[image]

Layer 6:
[image]

Support enforcer wasn't working when brim was enabled. It still was generated mid air.

### Project file & How to reproduce

[SimpleTreasureHeart.zip](https://github.com/prusa3d/PrusaSlicer/files/13666714/SimpleTreasureHeart.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0+win64

### Operating system

Windows 11

### Printer model

Prusa MK3+

axandar, 2023-12-13:
Here screens from new project and newly imported object wit brim.
Layer 1:
[image]

Layer 2:
[image]

Jan-Soustruznik, 2023-12-15:
Hello, @axandar,

thank you for report. We already know about this issue. Added into the main thread. Issue closed as duplicate.

#9601

Have a nice day

closed 2023-12-15 by Jan-Soustruznik