prusa3d/PrusaSlicer #11907: New 2.7.1 Installer trips Anti-Virus during update of icons

NWSpitfire, 2023-12-14:
### Description of the bug

During update installation of PS 2.7.1 the installer trips Norton 360 Anti-Virus and the install hangs (as Norton Has blocked it). Update still fails with Norton Disabled.
[image]
[image]
[image]

### Project file & How to reproduce

Have standard Norton 360 installed on PC with stock settings.
Download updated .exe through PS App when it notifies of new update.
Run .exe as administrator.
Click install PS 2.7.1 update (do not choose to update drivers too).
Antivirus will trip when installation reaches icons.
- Same steps with Norton Disabled.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.1+win64-202312121425_signed

### Operating system

Windows 11 Pro Build 22621

### Printer model

Custom CoreXY

NWSpitfire, 2023-12-14:
[image]

Norton AV Error message

yaronze, 2023-12-14:
Same problem on Windows 10 Pro.
Norton Anti Virus blocked all png files and installation fails.
I installed version 2.7.0 again that i can work again.

closed 2026-09-29 by prusaslicer-bot[bot]