prusa3d/PrusaSlicer #11908: Sync Feature for PrusaSlicer?

The-Best-Codes, 2023-12-14:
**Sync option to sync printer profiles across devices?**
I think it would be useful if I could use my prusa3d / printables.com account to sync my printer profiles and settings across devices. Just a suggestion! :D

neophyl, 2023-12-14:
Please search before opening new issues.  For example this one is easily found using the search term sync
https://github.com/prusa3d/PrusaSlicer/issues/10020
or how about https://github.com/prusa3d/PrusaSlicer/issues/9506
https://github.com/prusa3d/PrusaSlicer/issues/8891

Jan-Soustruznik, 2023-12-15:
Hello, @The-Best-Codes , 

thank you for your feature request. Yes, we already know about this request. I closed this issue as duplicate. 

Added into the main thread https://github.com/prusa3d/PrusaSlicer/issues/4898

Have a nice day

closed 2023-12-15 by Jan-Soustruznik