prusa3d/PrusaSlicer #11909: No Option to Disable Binary Gcode in 2.7.1

lucasoskorep, 2023-12-15:
### Description of the bug

I have prusa slicer 2.7.1 installed on linux. When I open it, it only outputs to binary gcode now which I in no way want. 

Going to print settings -> output options I am presented with no option to turn off the binary gcode feature. 
[image]

Please enable the ability to disable binary gcode.  This binary gcode breaks octoprint and several other programs I would rather use than the built in and increasingly forced proprietary options. 

### Project file & How to reproduce

Just open prusaslicer 2.7.1 on a linux box.  Tested on debian and Arch linux so it shouldnt be a repo packing issue. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Arch linux, and Debian 12

### Printer model

MK4

lucasoskorep, 2023-12-15:
Wait this was just moved to preferences -> other
[image]

Docs are still out of date for this so going to keep this issue open unless there is another way to file an issue for docs. 
https://help.prusa3d.com/article/binary-g-code_646763

hillsandales, 2023-12-15:
[Improvements fixed with respect to 2.7.0](https://github.com/prusa3d/PrusaSlicer/releases)

> The option Export as binary G-code was removed from Print Settings. Instead, there is a new option in Printer Settings named Supports binary G-code so it can be set at printer level. There is also a new global switch in Preferences->Other, which controls whether binary G-code will be generated for printers which support it. It is therefore easy to turn the feature on or off without doing any changes in profiles (https://github.com/prusa3d/PrusaSlicer/issues/11734, https://github.com/prusa3d/PrusaSlicer/issues/11873).

closed 2023-12-15 by Jan-Soustruznik