prusa3d/PrusaSlicer #11910: Make initial layer height configurable per filament, instead of Prusament PA11 specific g-code

timur-tabi, 2023-12-15:
### Description of the bug

The default starting g-code for 0.4mm and larger nozzles has this oddity:

```
{if filament_settings_id[initial_tool]=~/.*Prusament PA11.*/}
G1 Z0.3 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E9 F1000 ; intro line
{else}
G1 Z0.2 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E12.5 F1000 ; intro line
{endif}
```

My understanding is that the reason for this g-code that is specific to Prusament PA11 is that the recommendation is to use a glue stick with this particular filament, and so this code increases the Z height by 0.1mm to compensate.

This is silly.  A much better approach would be to include a filament-specific (and/or print-specific) option to add 0.1mm to the layer height if using a glue stick, so that this change can be used with any filament.  Then, just enable this option for the default Prusament PA11 configuration and leave it off for all other filaments.

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Ubuntu 22

### Printer model

MK3s

rtyr, 2023-12-15:
It does not change the layer height, it only changes purge line behavior.

closed 2023-12-15 by rtyr

timur-tabi, 2023-12-15:
@rtyr Closing this was premature.  Why does the PA11 need a higher purge line?

rtyr, 2023-12-15:
We are selling steel sheet for PA filaments. This condition is just to squish purge line less to this sheet. Glue does not play any role here and is not even used with this sheet.

ttabi, 2023-12-15:
That is still justification for making it a configurable option instead of this silly hack in the gcode.  What about other filaments and steel sheets?  What is so special about the combination of a steel sheet and PA filament that a slightly higher prime line is necessary?  And why don't you do the same thing with the 0.25mm nozzle starting g-code?

rtyr, 2023-12-15:
Its our filament and our sheet, we decided we will use different purge line for it. Standard purge line uses much higher flow and PA11 adhesion to PA sheet is very strong. 

Anyway, your initial request is still a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3209.

ttabi, 2023-12-15:
> Anyway, your initial request is still a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3209.

Thanks, I'll just upvote that issue instead.

rtyr, 2023-12-15:
> And why don't you do the same thing with the 0.25mm nozzle starting g-code?

It is carbon fiber filament, it cannot be printed with 0.25mm nozzle and it is not compatible with 0.25 printer profile.