prusa3d/PrusaSlicer #11912: Error to create the G-code

Alpine03, 2023-12-15:
### Description of the bug

G-code export to C:\Users\klaus\AppData\Local\Temp\.18412.gcode failed due to invalid custom G-code sections:
start_gcode
Parsing error at line 51: Variable does not exist
; M117 Print [input_filename]; Display: Printing started...
              ^
Please inspect the file C:\Users\klaus\AppData\Local\Temp\.18412.gcode.tmp for error messages enclosed between
        !!!!! Failed to process the custom G-code template ...
and
        !!!!! End of an error report for the custom G-code template ...
for all macro processing errors.

### Project file & How to reproduce

Slice File
[Slicer 2.7.1 Fehler.docx](https://github.com/prusa3d/PrusaSlicer/files/13683431/Slicer.2.7.1.Fehler.docx)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.1+win64-202312121425_signed

### Operating system

Windows 11

### Printer model

Canon TS6350

neophyl, 2023-12-15:
Please read https://github.com/prusa3d/PrusaSlicer/issues/11885

Jan-Soustruznik, 2023-12-15:
Hello, @Alpine03 

try to update the string name https://github.com/prusa3d/PrusaSlicer/issues/11885#issuecomment-1850093856

Closed as duplicate issue. https://github.com/prusa3d/PrusaSlicer/issues/11885

Thank you, have a nice day

closed 2023-12-15 by Jan-Soustruznik