prusa3d/PrusaSlicer #11913: Make binary GCODE an option selectable when saving file

mix579, 2023-12-15:
I like the bgcode option — definitely helped to address (partially) the slow file upload issue.

I also like that it has moved from print settings to printer settings.

However, while I usually prefer to create binary gcode, sometimes I want to create text gcode for debugging purposes.

My suggestion:

Add a selector to the file export dialog, such as a checkbox under the file list, to disable/enable bgcode on a temporary basis.

neophyl, 2023-12-15:
Can't you just use the global disable in preferences if you are debugging to turn it off ?

mix579, 2023-12-15:
Which is what I'm doing now but it's a lot more mouse clicks for a one off task.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena