prusa3d/PrusaSlicer #11914: Under extrusion or gaps at seams with latest versions of PrusaSlicer/firmware

FutureSharks, 2023-12-15:
### Description of the bug

Seams have gaps, holes or just look worse than before:
1. GOOD: Prusa Mini+ firmware 4.x+ Slicer 2.7.0x
2. NOT GOOD: Prusa Mini+ firmware 5.1+ Slicer 2.7.1, input shaper NOT enabled
3. BAD: Prusa Mini+ firmware 5.1+ Slicer 2.7.1, input shaper enabled

I can't tell if it's an issue with PrusaSlicer, my Prusa Mini+ firmware or the printer settings as all 3 of these were updated at once.

Here's some photos to show what I'm talking about, this is without any input shaper and compares old vs new versions:

[image]

And this compares input shaper vs no input shaper on new versions:

[image]

### Project file & How to reproduce

The model is [here](https://www.printables.com/model/637971-key-coin-and-small-things-tray) if it helps but I don't think it's important, I see it in all prints.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Mac

### Printer model

Prusa Mini+

vgdh, 2023-12-15:
I also have seem issues with kingroon printer on prusaslicer 2.7.1. I sliced the same file in cura and it was printed like a charm without any issues. I'm looking into trying older version to verify the issue if i have a time for it.

vgdh, 2023-12-15:
and what is strange, some of the seams are uglier that the others. But some are still very good.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

go4blossi, 2023-12-15:
Hey, i have the same problem after i updated my printer and try to use input shape

vazmuten, 2023-12-17:
I have same issue using Mini and think that this is a HUGE problem that should be fixed ASAP.

FutureSharks, 2023-12-17:
Could someone test with an older version of PrusaSlicer to rule out the firmware?

vazmuten, 2023-12-17:
Here are some results showing that FW 5.1.2 using Input Shaping causes the problem: 
[image]

arthurptj, 2023-12-17:
having same issue

carnas1, 2023-12-17:
having same issue 
[image]

jeloneal, 2023-12-18:
Same issue here. Underextrusion also on some surfaces after the swith to 5.x firmware.

D3nkis, 2023-12-25:
Hi colleagues, 

Same issue here. 
The previuos 4.x firmware was perfectly, but after I've switched to the new one, the quality along the seams is unacceptable.

dpvdberg, 2024-01-02:
Exactly the same issue here!

tysonsw, 2024-01-03:
Same issue here.
Prusa MINI+ with firmware 5.1.2 and using PrusaSlicer 2.7.1. 
Had almost no seams with the previous profiles.
The problem is present with both add:north PLA and PETG filament.

neophyl, 2024-01-05:
Lynn on the thread about this on the Prusa forum has brought up a good point.  Are those of you experiencing this issue using modified linear advance settings for your filament profiles ? As none of you have actually attached a saved PS project file to this issue its impossible to check that at the moment, hence the question.

I know from previous communication with tysonsw that the add:north filament profile he is using is not an IS profile but the standard one.  Might be worth trying the adjusted values from a IS adjusted profile as a starting point if they arent there already to see if it makes a difference.

FutureSharks, 2024-01-06:
> As none of you have actually attached a saved PS project

I can attach a project file if it helps but I assume not since it seems for most people a problem for every project/model that uses the newer IS profiles. 

> Might be worth trying the adjusted values from a IS adjusted profile as a starting point

How does one do that?

D3nkis, 2024-01-06:
Hi, 

the problem still exist on the MK4 printers since the same firmware version 5.1.2 and the Prusa Slicer 2.7.1

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3651

sirgrej, 2024-01-06:
After doing some tests with different MK4 FW versions I now thing this might be a slicer issue. I have tested fw 5.1.2, 5.0.1 and 4.7.2. The results look all the same independent from input shaping.

Maybe it is worth to try another slicer version or even another slicer to track the issue down

Mikyner, 2024-01-06:
[image]

D3nkis, 2024-01-06:
Is it printed with Input shaper speed profile in Orca?
Much better than PS.

Mikyner, 2024-01-06:
Yes, Orca Slicer contains only IS profiles for MK4
0.2mm Standar @MK4

D3nkis, 2024-01-06:
In afternoon I want to test the printing quality with Orca, I never used it before, I’ve choosed the Prusa Mini printer type, but I could’t find the IS profile and I gave up :D

Are you happy with this quality and will you switch from PrusaSlicer to Orca until Prusa doesn’t solve the problem?

Mikyner, 2024-01-06:
Unfortunately Orca does not include an IS profile for Prusu mini.
For now, I have no choice but to use Orca slicer, because I can throw away a lot of prints (from expensive filament) due to a bug in PS.

In Orca I miss a structural profile like in PS, I guess I'll have to create one because it's not available for download anywhere.

Anyway, if print profiles for MK4 were available for Orca (for example structural), or could be easily transferred from PS, I might stick with Orca, it has a lot of tweaks I like.

sirgrej, 2024-01-06:
I have tested prusaalicer 2.6.1 with mk4 and firmware 5.1.2 and the seam looks much better 

The upper print is with 2.6.1 and the lower is with 2.7.1

[image]

thenhustey, 2024-01-09:
For me it seems as general problem with under-extrusion on beginning (at least) perimeter.
[image]
[image]

The seam is under-extruded. And also this perimeter is under-extruded on its beginning.
[image]

I printed 2 copies of this part and both look exactly the same.

MINI+  
FW:  5.1.2
PrusaSlicer 2.7.1 +win64  
Nozzle: 06,mm  
Material: PLA  
Printer settings: 0,15mrn STRUCTURAL  
Filament profile: Prusament PLA  
Printer profile: Original Prusa MINI & MINI+ Input Shaper 0.6 nozzle

mimeister, 2024-01-10:
I'm seeing these quality issues regarding the seam on my Mini+, too. However, the biggest issue came from the K-factor for linear advance being incorrect, because the input shaping printer profile identifies as a different printer model (`PRINTER_MODEL_MINIIS `instead of `PRINTER_MODEL_MINI`), but some (not all) filament profiles that I migrated check for Mini without IS:
```
M900 K{if printer_notes=~/.*PRINTER_MODEL_MINI.*/ and nozzle_diameter[0]==0.6}0.12{elsif printer_notes=~/.*PRINTER_MODEL_MINI.*/ and nozzle_diameter[0]==0.8}0.06{elsif printer_notes=~/.*PRINTER_MODEL_MINI.*/}0.2{elsif nozzle_diameter[0]==0.8}0.01{elsif nozzle_diameter[0]==0.6}0.04{else}0.05{endif} ; Filament gcode LA 1.5
{if printer_notes=~/.*PRINTER_MODEL_MINI.*/};{elsif printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}M900 K200{elsif nozzle_diameter[0]==0.6}M900 K18{elsif nozzle_diameter[0]==0.8};{else}M900 K30{endif} ; Filament gcode LA 1.0
```
So the above code produces M900 K0.2 for my Mini without IS (correct value), but M900 K0.05 if the IS profile is used. Adding a simple `or printer_notes=~/.*PRINTER_MODEL_MINIIS.*/` fixes that (or just remove the check for the printer model at all, if you don't need it), which seems to mitigate the worst underextrusion.
But still, seams don't look good as others have pointed out before.

### Input Shaping with bad K
Really bad underextrusion at the seam. Somewhat expected because of the much too low K-factor.
[image]
### Input Shaping with correct K
At least no big gaps, but still some underextrusion.
[image]
[image]

carnas1, 2024-01-16:
Has anyone contacted support? if so, what was the conclusion?

FutureSharks, 2024-01-17:
> Has anyone contacted support? if so, what was the conclusion?

The [PrusaSlicer page](https://www.prusa3d.com/page/prusaslicer_424/) says this:

> Found a bug? Let us know on [Github](https://github.com/prusa3d/PrusaSlicer) or via [community forum](https://forum.prusa3d.com/forum/prusaslicer/).

Both of which I have done 😅

uncovery, 2024-01-18:
[image]

Same here, the seams are really terrible. My MK3 used to have much nicer seams.

carnas1, 2024-01-18:
good day. the problem has been reported for a long time. Is there really no reaction from Pruša?

uncovery, 2024-01-19:
ok, so I talked to Prusa support and they recom
[image]
mended to change the "External perimeters first" in the advanced settings of the "Layers and perimeters" section to ON.  I asked why this is not on by default and was told that it might affect overhangs. Anyhow switching this on had a dramatic effect for the better. I also did seams on "REAR" instead of aligned and they came finally in a straight line. Seems I have to do 2 different presets for "nice" and "overhangs"?

mix579, 2024-01-19:
Interesting. So if you have overhangs, you're still screwed...

webbes, 2024-01-20:
Same issue on my Mini+ here. Huge gaps at the seams. Does not matter if I select the IS printer profile or not.  I'ld never expected Prusa to deliver such a terrible slicer or firmware version. I'm on stock Prusa, using Prusament because I don't want to deal with quality issues. They used to just work. Attached picture is with latest firmware (5.1.2) and Prusa Slicer (2.7.1), non IS printer profile, Prusament PETG filament. Only change from default is 30% infill instead of 15% and a Diamondback 0.6 nozzle.

[image]

andrewmeyer, 2024-01-21:
Its not unique to Prusa firmware, I am encountering the same issue on a klipper printer, I'm hoping there is a fix soon.

Another post seemed to suggest setting "extra length on restart" to 0.06 (0.15*0.4)  but this looks like it only applies when a toolchange is performed, not on a move

bszonye, 2024-01-26:
@mimeister I'm seeing this problem with the `Prusament PLA @MINIIS` profile, which does not use M900 linear advance gcode, only M572 pressure advance:

    M572 S{if nozzle_diameter[0]==0.6}0.17{elsif nozzle_diameter[0]==0.8}0.12{elsif nozzle_diameter[0]==0.4}0.3{elsif nozzle_diameter[0]==0.25}0.85{else}0{endif}

The M900 gcode that you quoted does appear in the generic `Prusament PLA` profile, but that one excludes MINIIS in the compatible printers condition, so normally it's not selectable for input shaping, only the `@MINIIS` version that omits the M900 command.

That makes me wonder whether it would help to add the M900 command, or whether it would just conflict with M572. I don't know enough about the two systems to know whether they work together, whether one is preferable to the other, or whether the M572 settings are tunable.

I might try switching to external perimeters first, as I'm not as worried about overhangs as these terrible seams.

bszonye, 2024-01-26:
I changed the M572 S-value from 0.3 to 0.2 and got better (but still bad) results. So with that in mind, I used the [linear advance calibration tool](https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/pattern_method.html) suggested in the forum thread to find a better K-value. I also made a version of the calibration gcode that used M572 S-values instead of M900 K-values and found that they had very similar effects at the same values.

Here's my test print at three different S-values:
[image]

The brown print uses the default `M572 S0.3` setting from the stock Prusament PLA configs, and in person it looks torn open. The S0.2 seam has a much narrower gap, but it's still unacceptably bad. The S0.17 seam actually looks better in person than it does in the photo. There's no gap at all, only a shallow crease.

bszonye, 2024-01-27:
After sleeping on it, I have some more thoughts on what's happening and why it's so much worse with Input Shaping.

In learning about linear advance and pressure advance, I found that the M900 K-factor for linear advance has the same units as the M572 S-factor for pressure advance. According to [this M900 reference](https://reprap.org/wiki/G-code#M900_Set_Linear_Advance_Scaling_Factors) they used to have different scaling factors based on extruder gearing, but in newer Marlin versions, K and S should be equivalent.

I tested that using [Ellis' Pressure Advance / Linear Advance Calibration Tool](https://ellis3dp.com/Pressure_Linear_Advance_Tool/) to create a calibration print with M900 codes, then I derived a second test print with M572 for an apples-to-apples comparison. I got basically identical results from both tests, so it does look like M900 K-factor and M572 S-factor are interchangeable on the Mini.

Thus I find it strange that the default settings for Prusament PLA (at 0.4mm) use K0.2 for the general profile and S0.3 for the IS profile. In my calibration prints, 0.2 was already a bit too high, and 0.3 had significant gaps and other artifacts. I suspect that's why people are mostly seeing this problem with IS enabled: the IS profile has the S-factor set much too high *and* the higher print speeds exaggerate the problem.

joachimklug, 2024-01-28:
I do have the same issue on my Prusa MK4 which is running on Firmware 5.1.2. My files are sliced with Prusa Slicer 2.7.1 and the latest configuration.

In order to reproduce the issue I have created a test file. It is a cylinder with 2cm diameter and 0.8mm wall thickness. The .3mf file as well as the sliced .bgcode file can be downloaded [here](https://drive.google.com/drive/folders/1a1Va4oZiOkJa5gcItdY9dl5a4wptPiqC?usp=sharing). The slicing was done with stock settings for Prusa Slicer:
- 0.2mm QUALITY layer height
- Prusament PLA
- MK4 0.4mm no IS
- the cylinder was rotated before slicing in order to have the seam in the front for the video.

As a result there is the quite ugly seam which shows characteristics of under-extrusion. I have created a [slow motion video](https://youtu.be/4wioVQlX5Vw) and uploaded it to YouTube to show the result and to locate the issue. In the video you see that at the start of a line the printhead is in the right location but filament extrusion only starts AFTER the printhead is moving. Thus leaving a gap without filament which forms the seam. On the other side (on line end) the extrusion of filament is going until the very end. This can be confirmed by following macro image. On the left side you see the end of the line which somehow forms a straight line. However on line start (on the right side) the line forms a kind of zigzag which is where the filament is not ready to extrude but the printhead already moves. What can be seen as well is that on the very first layer (bottom) the extrusion starts on the right spot. So the extrusion starts way earlier than on all other layers.

[image]

I have contacted Prusa Support twice and got following answers which try to mitigate the issue:
- Raise extrusion temperature: Therefore I have sliced the pipe into four 1cm slices and increased temperature every 1cm. 215 -> 225 -> 235 -> 240. Only on 240 the seam was greatly reduced. I think this is due to the fact that the filament is so fluid now that it starts to drip at the right point to fill in this gaps. I don't consider this a real solution. I've uploaded an image of the results.
- Clean nozzle and ensure extruder/nozzle is tightly secured: I did both. First checked all screws on the extruder and nozzle (which were already tight) and then did a cold pull. After the cold pull a small print looked already better. I did a cold pull again to make it even better and run the test print after that. The issue was the same as before. No improvement.
- Increate FlowRate: I didn't check this so far.

[image]

There is no conclusion or result from my side so far. Just wanted to share my findings.

vivienm, 2024-01-28:
I observe the same behavior on my MK4 (FW 5.1.2).

* Prusament PLA;
* No IS, 0.2 mm quality profile;
* All stock settings.

Changing the retraction length or speed doesn't seem to have a significant impact. I've noticed a couple of things though, that may or may not be relevant:

* Seam is better when using the classic perimeter generator instead of Arachne;
* Seem is also better when disabling linear advance (that is, setting the K parameter to zero).

I've checked my LA calibration (using [this tool](https://garethky.github.io/PrusaSlicerPressureAdvanceCalibration/)) and the default value of K = 0.05 seems correct, so I'm not sure how to interpret this result.

For now, I just set the deretraction length to 0.06, which seems to give the best results in my tests.

LuckyResistor, 2024-01-30:
---

I noticed large gaps or filament blobs at the seams when printing on the Prusa XL with 0.6mm or 0.8mm nozzles, using input shaper and the latest firmware and slicer. I used fresh spools of Prusament PETG Signal White for all the prints shown. For scale, this print is large. The planter shown is 200mm in Z, 320mm in X, and 220mm in Y.

I have attached the PrusaSlicer project and gcode that led to the results shown. The defects worsen with random seam positions. I used the "0.4mm QUALITY" profile.

- [Print 0_8mm.zip](https://github.com/prusa3d/PrusaSlicer/files/14100150/Print.0_8mm.zip)

Just after the seam, a large gap, 10-15mm long, opens:
[image]

Here is a detail of the gap:
[image]

Here is the same issue on the inside of the print.
[image]

Interestingly, this was not the only problem. I also noticed over-extrusion of the infill, which led to very uneven layers and a build-up of material on the nozzle.

I tried the same print again with a 0.6mm nozzle, first with the "0.32mm STRUCTURAL" profile, which caused so many problems that I had to stop the print. Then I tried the "0.32mm SPEED" profile, with which I could complete a full print.

With random seam positions, the effects were different. Instead of gaps, I got blobs of filament that led to defects up to four layers above the seam. Also, the print was visibly uneven with under and over-extrusion along the sides.

[image]
[image]

So I changed the random seam positions back to aligned seams and used draw on seams to put them in a good place. But this led to even worse effects on the print. The problems were apparent already while printing:

[image]

After the print, the seam had not only the usual visible dent but also a large blob of material after the seam in the print direction.

[image]
[image]

The strangest thing is, the next day I printed the plate for the flower pot, using the same filament and the same profile (0.32mm SPEED), and I got a much better result.

joachimklug, 2024-01-30:
Here are some more tests and its results. Printed in Prusament PLA with default slicer profile, 0.4mm nozzle and 0.2mm layer height.

[image]

1. MK4 - Firmware 5.1.2 - Prusa Slicer 2.7.1 - Prusa Slicer Settings 1.11.11 - Disabled Arc fitting - Result: unchanged
2. MK4 - Firmware 5.1.2 - Prusa Slicer 2.7.1 - Prusa Slicer Settings **1.7.2** from April 2023 - Result: unchanged
3. MK4 - Firmware 5.1.2 - Prusa Slicer **2.6.1** - Result: unchanged
4. MK4 - Firmware **4.7.2** - Prusa Slicer 2.7.1 - Result: unchanged
5. MK4 - Firmware **4.7.2** - Prusa Slicer **2.6.1** - Result: unchanged
6. **MK3S** - Firmware 3.13.2 (latest) - Prusa Slicer 2.7.1 - Result: Seam looks much better. Some minor defects in the bottom half
7. MK4 - Firmware 5.1.2 - Prusa Slicer 2.7.1 - Deretraction extra length: 0.06 - Result: Seam is less visible but gap is still there
8. MK4 - Firmware 5.1.2 - Prusa Slicer 2.7.1 - Deretraction extra length: 0.08 - Result: Seam is almost gone. Print looks very good
9. MK4 - Firmware 5.1.2 - Prusa Slicer 2.7.1 - Deretraction extra length: 0.10 - Result: Seam is almost gone. Maybe even a bit too much. Print looks very good

The conclusion to my test: The print quality of the seam with standard profile is much better with the MK3S. For the MK4 the ugly seam (gap) is showing in any configuration of slicer and firmware version. This lets me assume that this behaviour was there from the beginning. The setting for detraction somehow solved the issue which also was shown in my last post. (Filament extrusion only starts after printhead moves)
However the question in the first place is: Why is the filament not ready to be extruded? While it is ready on my MK3S?

RcrossF, 2024-02-02:
Same issue here on my MK3S, running a revo with 0.6mm nozzle, stock profiles (although my usual ones do the same). Adding ~0.2mm of extra restart length has mitigated the issue but it's still somewhat inconsistent at the seams

mimeister, 2024-02-06:
Stumbled upon [this issue](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3546) from November in the buddy firmware repo yesterday. That probably describes the very same problem. Just mentioning it here to get the issues linked.

mimeister, 2024-02-06:
So I dug a little bit deeper into the linear advance vs. pressure advance topic because of @bszonye's posts.

> I tested that using [Ellis' Pressure Advance / Linear Advance Calibration Tool](https://ellis3dp.com/Pressure_Linear_Advance_Tool/) to create a calibration print with M900 codes, then I derived a [...]

Turns out these results are very expected, because in fact the buddy firmware does not interpret M900 as linear advance anymore, but instead calls M572 internally. This was stated in [[QUESTION] Latest MK4 firmware seems to have Linear advance and Pressure advance?](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3340) and I did confirm this reviewing [the code](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/lib/Marlin/Marlin/src/gcode/feature/advance/M900.cpp). So basically this means, M900 K\<value> and M572 S\<value> can be used interchangeably at the current state of the firmware, both will configure pressure advance.

> Thus I find it strange that the default settings for Prusament PLA (at 0.4mm) use K0.2 for the general profile and S0.3 for the IS profile. In my calibration prints, 0.2 was already a bit too high, [...]

That is strange, indeed. Also while tuning PA with [@garethky's tool](https://garethky.github.io/PrusaSlicerPressureAdvanceCalibration/) I found 0.3 to be much too high for my Mini+, too (using some silk PLA).
But still I cannot get rid of the underextrusion issues when deretracting with just tuning PA. I also tried adding some extra length on deretraction, but even 0.12mm does not fix the gap.
However, adding extra length on deretraction can only be a temporary workaround. Looking at the pictures above, the underextrusion seems to be quite inconsistent and adding extra material to close the gap reliably will probably lead to blobs and bulges where the underextrusion is less pronounced.

Please, Prusa Research, have a look into this. It's an unacceptable print quality issue and there are at least three related open issues on Github now with no reaction whatsoever since end of November. Maybe tagging Josef himself helps to draw some attention?? @josefprusa

ParaDiddle777, 2024-02-08:
This is unacceptable on my near $4000 XL too. I can't do anything to get rid of gaps in my seam on a single walled print.

Benigans, 2024-02-08:
> I changed the M572 S-value from 0.3 to 0.2 and got better (but still bad) results. So with that in mind, I used the [linear advance calibration tool](https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/pattern_method.html) suggested in the forum thread to find a better K-value. I also made a version of the calibration gcode that used M572 S-values instead of M900 K-values and found that they had very similar effects at the same values.
> 
> Here's my test print at three different S-values: ![seams](https://private-user-images.githubusercontent.com/5016349/300059979-487f7740-7ecf-4fc0-8adc-82d4a245f655.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc0MjgyODEsIm5iZiI6MTcwNzQyNzk4MSwicGF0aCI6Ii81MDE2MzQ5LzMwMDA1OTk3OS00ODdmNzc0MC03ZWNmLTRmYzAtOGFkYy04MmQ0YTI0NWY2NTUuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDhUMjEzMzAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2VmZGZmMGE2MTc5MTE0OTQ5NmE2YWM2MjgwNDllNGFmMWM3ZjBmOTg2MTc0ZDQ2Y2M3YWNlMGJkZWExMDRkZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.XZh7S0s2N9NfShomWe4k5bAhOwuLdLkcFX-bwwQ0P2M)
> 
> The brown print uses the default `M572 S0.3` setting from the stock Prusament PLA configs, and in person it looks torn open. The S0.2 seam has a much narrower gap, but it's still unacceptably bad. The S0.17 seam actually looks better in person than it does in the photo. There's no gap at all, only a shallow crease.

Thank you for that !! I tried to change that setting with the PETG profile (value is 0.4 by default, set it to 0.23 instead) and the seam is much better now !
Tried with some prints and it looks a lot better now !

michal-repo, 2024-02-11:
@Benigans 
> Thank you for that !! I tried to change that setting with the PETG profile (value is 0.4 by default, set it to 0.23 instead) and the seam is much better now ! Tried with some prints and it looks a lot better now !

Where do you change that?

D3nkis, 2024-02-11:
> @Benigans
> 
> > Thank you for that !! I tried to change that setting with the PETG profile (value is 0.4 by default, set it to 0.23 instead) and the seam is much better now ! Tried with some prints and it looks a lot better now !
> 
> Where do you change that?

Hi Michal, 

Here it is:
[image]

jeloneal, 2024-02-11:
Custom Gcode on filament tab. There should be some code for each diameter of the nozzle.

ParaDiddle777, 2024-02-11:
Just to let everyone know. I upgraded to the 2.7.2 Alpha2 version and I think that fixed it. Try it out.,

michal-repo, 2024-02-11:
If issue is in config they can push fix in `Configuration update`

mixer3d, 2024-02-11:
no, it's not related to the config, but how the slicer cut the model

LuckyResistor, 2024-02-11:
I think fixed the problem with filament blobs, I also had – there are definitely less filament blobs in the print. The problem with the holes at the seam still persists.

[image]
[image]

- Printer: PrusaXL 5T IS
- The print was sliced using PrusaSlicer 2.7.2-alpha2 with the latest configuration updates.
- Default seam settings.
- Filament: Formfutura EasyFil™ ePETG Traffic White - which is even more forgiving then Prusament PETG
- Nozzle: 0.6mm
- Profile: 0.32mm Speed

[gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/14234034/gcode.zip)

I did another print, where I manually placed the seams approximate 20mm apart. The short travels helped a lot to create better seams. Yet, as you can see, as soon the travel distance is larger (at the very bottom), holes appear.

- Printer: PrusaXL 5T IS
- The print was sliced using PrusaSlicer 2.7.2-alpha2 with the latest configuration updates.
- Manually placed inner/outer seams, approximate 20mm apart.
- Filament: Formfutura EasyFil™ ePETG Traffic White - which is even more forgiving then Prusament PETG
- Nozzle: 0.6mm
- Profile: 0.32mm Speed

[image]
[image]

[gcode2.zip](https://github.com/prusa3d/PrusaSlicer/files/14234052/gcode2.zip)

michal-repo, 2024-02-11:
[image]
[image]
[image]
[image]
[image]
[image]

Two perimeters, 0.6 nozzle, 0.35mm, Generic PLA
1. PrusaSlicer 2.7.1
2. PrusaSlicer 2.7.1 Input Shaper
3. PrusaSlicer 2.7.2-alpha2, imported config
4. PrusaSlicer 2.7.2-alpha2, fresh config
5. PrusaSlicer 2.7.2-alpha2 Input Shaper, fresh config

I don't see any improvements.

Benigans, 2024-02-12:
> ![IMG_2572](https://private-user-images.githubusercontent.com/2189840/303949637-76908ab9-7339-4aad-8fdc-7100cb5045bb.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NDA5MzcsIm5iZiI6MTcwNzc0MDYzNywicGF0aCI6Ii8yMTg5ODQwLzMwMzk0OTYzNy03NjkwOGFiOS03MzM5LTRhYWQtOGZkYy03MTAwY2I1MDQ1YmIuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTJUMTIyMzU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2JlZGYxYWZjODhhMjY4ZDMzMzI3NDBjYTYzYTZkMTk3YmU0OWEzZGFjYTM4Yjk1YmZkZWNkMzM5MjgzYTI5MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ywFByBRUAWb5BLLo7n_ibzqjqhcDSP4YKCR2gKMnoTA) ![IMG_2571](https://private-user-images.githubusercontent.com/2189840/303949641-d502567e-f2f3-4836-b6e0-c33bd9f31ed5.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NDA5MzcsIm5iZiI6MTcwNzc0MDYzNywicGF0aCI6Ii8yMTg5ODQwLzMwMzk0OTY0MS1kNTAyNTY3ZS1mMmYzLTQ4MzYtYjZlMC1jMzNiZDlmMzFlZDUuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTJUMTIyMzU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDc4MTlhOTkwMzc4YWIxNmFkYTAzNjdiMjM2NWNjZmQ3NzdiY2JhMDYyYjIxZTJlODliNDZjNDI0OGMxM2RiZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.UkpyuGLE4zkmOPUp1NUJh_XiyQMPr0EZsEolO8gEYfc) ![IMG_2573](https://private-user-images.githubusercontent.com/2189840/303949646-6d6dcd2f-f851-42c3-8a6c-e33216cc4bb1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NDA5MzcsIm5iZiI6MTcwNzc0MDYzNywicGF0aCI6Ii8yMTg5ODQwLzMwMzk0OTY0Ni02ZDZkY2QyZi1mODUxLTQyYzMtOGE2Yy1lMzMyMTZjYzRiYjEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTJUMTIyMzU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yjk0NzI2Mzg4MDBiYjFhOGE4ZDBkMmU5YjM5MWMzNWFiNWU3N2RiZTNmZGNiNWRkMzMxZjM5ZjBiNzM2YWZmMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.VFk7Woh_3ppeEulao_qX_OMJan4IDb3cgHKx0UQGUKA) ![IMG_2576](https://private-user-images.githubusercontent.com/2189840/303949651-84514f20-016d-4a47-9bf0-7ecd8216e17a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NDA5MzcsIm5iZiI6MTcwNzc0MDYzNywicGF0aCI6Ii8yMTg5ODQwLzMwMzk0OTY1MS04NDUxNGYyMC0wMTZkLTRhNDctOWJmMC03ZWNkODIxNmUxN2EuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTJUMTIyMzU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzllMzg3NjFiNTkzNzllZmQ2MTQzZGYwNDE4NDU0NDJmYWFiNmI0NDZkOTI4ZDBmYWZjMzE0MWZlMGY0ZmU1YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.BNPztHGbEVCndIGORcxtQcyYvwF_VBNDLHCKF8hp4_s) ![IMG_2574](https://private-user-images.githubusercontent.com/2189840/303949654-013cee1f-fc6c-4e87-9cb5-4c02f8fa4004.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NDA5MzcsIm5iZiI6MTcwNzc0MDYzNywicGF0aCI6Ii8yMTg5ODQwLzMwMzk0OTY1NC0wMTNjZWUxZi1mYzZjLTRlODctOWNiNS00YzAyZjhmYTQwMDQuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTJUMTIyMzU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmZmYTA0Nzk1ZjhjZGUzMDhkZTVhMGI3NTQ2YTg1ZTc4NjMxNThmY2M5ZmZkOWFmYmQxNjEzYTUyMDFiNTRmYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.IDxQTHVwnLC0S4XbL2ib9N-89KWHMi5LUlZ7eWw3S50) ![IMG_2575](https://private-user-images.githubusercontent.com/2189840/303949661-925acaf9-82e3-43ac-acc1-2246e05f19ca.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NDA5MzcsIm5iZiI6MTcwNzc0MDYzNywicGF0aCI6Ii8yMTg5ODQwLzMwMzk0OTY2MS05MjVhY2FmOS04MmUzLTQzYWMtYWNjMS0yMjQ2ZTA1ZjE5Y2EuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTJUMTIyMzU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGMxZDFkZTM2NWI2NmM3NTliZjc2ZWY1MTEzZTRhMTFkYTlmMTUwYzc5NTk1NjFkMjAwMzBiZjUxMzMwMGI0ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.j2KbEuUV8uzabaAmVCrMi4iX8X-47e89SsaR7uAdUts)
> 
> Mini+, Two perimeters, 0.6 nozzle, 0.35mm, Generic PLA
> 
> 1. PrusaSlicer 2.7.1
> 2. PrusaSlicer 2.7.1 Input Shaper
> 3. PrusaSlicer 2.7.2-alpha2, imported config
> 4. PrusaSlicer 2.7.2-alpha2, fresh config
> 5. PrusaSlicer 2.7.2-alpha2 Input Shaper, fresh config
> 
> I don't see any improvements.

Could you share your STL ?

michal-repo, 2024-02-12:
[Shape-Cylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/14238638/Shape-Cylinder.zip)

this is not stl but shapes from prusa slicer.

michal-repo, 2024-02-12:
I just made same test with PrusaSlicer 2.6.1
My Mini is running on firmware version 5.1.2+13478

[image]

[image]

Are we confident that's Slicer issue and not firmware?

As a note, to run older slicer and don't mess up with latest config use datadir option:
`prusa-slicer.exe --datadir [...]\config\`

Benigans, 2024-02-12:
I’ll try your file on my printer when I get back home

carnas1, 2024-02-15:
good day. this issue has been open for two months. is it the case that no solution to this problem has been found in that time? Or is there a setting that works? why are you silent?

Benigans, 2024-02-16:
Hi, I tried to print @michal-repo file but nothing interesting on my printer. It printed ok with default settings.
I printed an other file to see the improvements when changing the M572 value.
The default for prusament PETG, 0.4n is 0.4. I also made tests with 0.23. See the pictures below :
Top is 0.23, bottom is 0.4. Huge difference
[image]

on the inside: 
0.4
[image]

0.23
[image]

I received the bondtech extruder and installed it today. The prints above are after the bondtech extruder installation. It was the same with stock extruder. I didn't bought it for that issue but wanted to test the before/after just in case.

FutureSharks, 2024-02-18:
Folks, I don't think we need more photos.

FutureSharks, 2024-02-18:
Does anyone know how we can actually get support to fix it?

[167 more replies]