prusa3d/PrusaSlicer #11915: Deretraction Extra Length not always applied

PascalGodin, 2023-12-15:
### Description of the bug

Deretraction Extra Length will not always be applied. Tried on multiple printer configurations and print settings but will still randomly apply the extra length. Noticed that changing the infill or the top/bottom layer quantity will change the positions of where it will apply or not the extra length but it still seems random. 
Applied an extra length of 10mm to be easy to spot.
This is issue was also reproduced in 2.6.1
The reason for using an extra length is otherwise it will leave a gap after retractions/travels. This is more noticable with aligned seams and will be more pronounced the bigger the nozzle. It is usually hidden with solid infill but for parts without infill and especially with only one perimiter this is critical. Usually a value of 0.025 to 0.1 will close the gap. 

[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[Extra Length Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/13686329/Extra.Length.Issue.zip)
Slice and look for deretraction values after travel moves in gcode preview

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Windows 10

### Printer model

Prusa XL,  MK4

RoadrunnerJDD, 2024-01-16:
I am having the same issue.  I am fairly new to PrusaSlicer and this was my first time switching to a 0.8mm nozzle after the switch from cura.  I recognized the problem and started adjusting the Deretraction Extra Length setting but prints were not getting better.  After examining the G-code I found that no matter what I set the Deretraction Extra Length setting to it always only deretracted the retraction length alone.

Version of PrusaSlicer
Version 2.7.1+win64

Operating system
Windows 10

Printer model
Creality Ender 3 S1

RoadrunnerJDD, 2024-01-18:
Found a workaround.  At least in the tests I have done if I set the Retract Amount Before Wipe to any percentage value other than 0% then the Deretraction Extra Length works as expected.

[image]
[image]

timangus, 2024-11-18:
Are any of you using "Avoid crossing perimeters"? I've noticed that if you have it enabled, the deretraction is applied, but *before* the moves within the perimeters, defeating the point.

hdiethelm, 2025-02-10:
Just a year old bug and it is still there...
I found an other workaround. If you typical GCode for deretraction is G1 E1 F2100, use:
Print Settings -> Output Options-> G-code supstiutions:
Add Find: **G1 E1 F2100** Replace with: **G1 E1.1 F2100**
And you just added 0.1mm of deretraction.

neophyl, 2025-02-10:
Alternative you could use the extra length on restart setting as described above instead of using a find/replace to add the extra .1 on a de-retract.  Not that that is a fix for a **missing** detract anyway as there wouldn't be any command to add an extra amount to in the first place.

Wegerich, 2025-04-21:
Looks like I am having the same problem, amongst other problems with wipe.
Deretraction is silently blocked by the wipe feature in my config. 
https://github.com/prusa3d/PrusaSlicer/issues/14475

I found an easy workaround was to disable wipe, I didn't lose any print quality.