prusa3d/PrusaSlicer #11917: Crash - paint-on supports and reload from disk (2.7.1)

the-ress, 2023-12-15:
### Description of the bug

PS crashed when I tried to remove paint-on supports after reloading a STL from disk.

Crash dump: [prusa-slicer.exe.23392.zip](https://github.com/prusa3d/PrusaSlicer/files/13690015/prusa-slicer.exe.23392.zip)

### Project file & How to reproduce

1. I had a model opened in PS and used 'paint-on supports' to block supports from a part of it
2. I made some tweaks in Fusion 360 and re-exported the model
3. I right-clicked the model and selected 'reload from disk'
4. The painted 'block supports' areas survived the reload (I expected them to be removed), but they didn't seem to affect the generated supports
5. I tried painting some more 'block supports' areas, but it was behaving weirdly
6. So I clicked 'remove all selection'
7. PS immediately crashed

Now that I think of it, the red 'block supports' areas might have had the shape of the model before reload.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Windows 11 (22631.2787)

### Printer model

Prusa MK4

the-ress, 2023-12-15:
I think I managed to reproduce it: [prusa-slicer.exe.27452.zip](https://github.com/prusa3d/PrusaSlicer/files/13690098/prusa-slicer.exe.27452.zip)
(This time it crashed during painting more supports after reload.)

Jan-Soustruznik, 2025-04-03:
Hello, @the-ress,

Thank you for your report, but we are unable to reproduce the reported issue using your repro steps. We tested it with version 2.7.1, which you mentioned in the issue description, as well as the latest release, 2.9.1, and got the same results.

Additionally, we made some enhancements to multimaterial painting in 2.9.0-Alpha1, so it is possible that the issue you reported has already been fixed.

Could you take a time to check if the problem still persists on your side?

Thank you, and have a great day!

closed 2026-09-29 by prusaslicer-bot[bot]