prusa3d/PrusaSlicer #11921: Brim height and chamfered brim options feature request

EducatingSavvas, 2023-12-16:
I'd like to request/suggest the feature or option to increase brim layer heights - within the advanced settings > print settings > Skit and brim > Brim options.  At the moment the brim is only 1 layer thick. I understand I have the option to add a shape - cylinder, and align the shapes using the object manipulation coordinates, however this is a lengthier process and doesn't form a bespoke shape around the first layer perimeter of the object. Maybe an alternative would be the option to add a shape based on the first layer permitter of the object selected. Another feature I'd like to see is the ability to chamfer the brim into the object. Both options are to allow for unusual and taller objects to be printed with better print adhesion support, while the brim remains as a permanent feature of the printed object and aids display and preventing tipping. This is partially aesthetic and practical, and would increase the versatility of the Prusa slicer by removing the requirement to use other third party software.

[image]

Tupson444, 2026-09-15:
still relevant

denesius, 2026-09-17:
Amen!
If it's not addressed in 3.0.0 (which I haven't had a chance to eval), I hope it gets attention

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #15946.

closed 2026-10-21 by Jan-Soustruznik