prusa3d/PrusaSlicer #11922: File incomplete message when i switch to PS 2.7.1

ZedPrg, 2023-12-16:
### Description of the bug

"File incomplete" is shown on Printer LCD
Asks "Continue Anyway ?" - I proceeded and the printing was completely normal. I have tried exporting it several times and always with the same result.

### Project file & How to reproduce

[filamentclip_0.3mm_ASA_MK3SMMU2S_41m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/13692738/filamentclip_0.3mm_ASA_MK3SMMU2S_41m.gcode.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1+win64

### Operating system

Windows 11

### Printer model

Prusa i3 MK3S+ MMU2S

johngalt01, 2023-12-17:
i'm seeing the same issue suddenly occuring. i just updated all my firmware(main and mmu2) and newest slicer 2.7.1 for my MK3S+ MMU2

JeffAGreen, 2023-12-26:
I am also getting the same issue. Updated to 2.7.1 and using MK3s+ and MMU3 with latest firmware. Working fine on the MK4.

rtyr, 2024-01-01:
It needs to be fixed on the firmware side. See https://github.com/prusa3d/Prusa-Firmware/issues/4546#issuecomment-1863879349.

closed 2024-01-01 by rtyr