prusa3d/PrusaSlicer #11924: 2.7.1; Can't Change 2nd Color when adding a color change

mahornak, 2023-12-16:
### Description of the bug

I'm printing something that needs a color change.  I select the required layer height and add the color change.  Then I go to the first color and sucessfully change it to my first color.  However, when I right click on the 2nd color in the slidebar, 2.7.1 just freezes.  I'm not able to change the 2nd color to the color I want.  When I hit ESC, it unlocks.

### Project file & How to reproduce

[TED_LASSO_BELIEVE_no_loop_0.15mm_PLA_MINI_8h18m.zip](https://github.com/prusa3d/PrusaSlicer/files/13693386/TED_LASSO_BELIEVE_no_loop_0.15mm_PLA_MINI_8h18m.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10 Home, V. 22H2, OS Build = 19045.3803

### Printer model

Prusa Mini +

YuSanka, 2023-12-22:
Thanks for your report. But I can't reproduce your issue. 
A short video or more detailed steps to repro would be very helpful.
Nevertheless, I don't understand how could you change a first color from color bund, when it's impossible action. We don't allow to change a first color. First color in bund is always filament or extruder color.

mahornak, 2023-12-30:
Oleksandra,

Hello.  I've attached a screen grab of the online manual for PrusaSlicer.
In this section it tells you how you can alter a color for a new color
change that is added.  I can add color changes easily enough.  However, the
colors produced are standard and not what I really want.  So, I right click
anywhere in the new color bar and the software freezes.  I have to hit ESC
to unfreeze PrusaSlicer,  I can change the first (bottom) color by
selecting the filament color block but not the others (see attachments).
Please help me.  Thanks.

Mark

YuSanka, 2024-01-05:
@mahornak , unfortunately, your attachments wasn't added.
Please, resend them.

mahornak, 2024-01-05:
Oleksandra,

Here are the photos again.  Thanks.

Mark

YuSanka, 2024-01-10:
@mahornak But still nothing :)

mahornak, 2024-01-10:
Oleksandra,

This is very frustrating.  I've attached photos to the last 2 emails I sent
you.  You must have an anti-virus program that is deleting them. I've added
them to Google Drive and sent them to your email address.  I've also added
the link here.

https://drive.google.com/drive/folders/18ANLrk2dkRDjJKOh_s-tTx2clStXcsiN?usp=drive_link

I will also try pasting the photos in this email below.

[image: PrusaSlicer - color change problem.jpg]

[image: PrusaSlicer - cannot modify color change color.jpg]

Hopefully one of these attempts of getting you the photos will work.
Please let me know.  Thanks,

Mark

Jan-Soustruznik, 2024-01-16:
Hello, @mahornak , 

this is limitation of the github web page, I think. I tried use the google drive link, but we dont have a permission to download. Could you change the permission, please. Thank you

mahornak, 2024-01-16:
I've shared an item with you:

PrusaSlicer -  color change problem.jpg
https://drive.google.com/file/d/1TqtaJPDknatYsIEDy58sMNAoeEhDV0iW/view?usp=sharing&invite=CPuAjOoO&ts=65a6d2e2

It's not an attachment -- it's stored online. To open this item, just click  
the link above.

mahornak, 2024-01-16:
I found your access requests to my Google Drive folder and granted access.
Hopefully you can see my documents now.  Thanks for your patience.

Mark

mahornak, 2024-02-19:
Thanks for fixing the color selection bug in PrusaSlicer!  I see that the
color selection menu was opening up off of the screen.  It works now.
Thanks again.

Mark

closed 2026-09-29 by prusaslicer-bot[bot]