prusa3d/PrusaSlicer #11925: STL Seems to be In Meters

ciley417, 2023-12-16:
### Description of the bug

Suddenly I started getting this message every time I load a STL file:

The dimensions of the object from file Bracket (5).stl seem to be defined in
meters.
The internal unit of PrusaSlicer is a millimeter. Do you want to recalculate the dimensions of the object?

I am using Onshape Online CAD program, and my unit of measurements are set to Millimeters.
I have been using it for long time now. 
This started just recently when I was already using Prusa Slicer 2.7.0 version and it still happens 
with the latest version and it appears every time.

ciley@verizon.net

### Project file & How to reproduce

On loading a STL file.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0 and 2.7.1

### Operating system

Windows 11 Version 22H2.

### Printer model

ENDER 3 S1 and ENDER 3 S1 Plus

neophyl, 2023-12-17:
How about attaching the zipped up stl ?  That way it can be loaded to various different software to check out how they interpret its scale.

sbussinger, 2023-12-23:
I think this is a bug in OnShape. I'm having the same problem. If you export the part from OnShape with the units set to Millimeters it seems to work. If you set the units to Meters, it gets the message you indicated. It didn't act this way prior to a week or two ago. When I viewed the STL in a viewing program a test part showed the dimensions as 0.01 mm instead of the actual 0.01 m (10mm) so I'm guessing the units in the STL are messed up (I know nothing about the STL file format).

Here's a test STL file (a 10mm x 10mm x 10mm cube) in a ZIP file because STL files aren't allowed:
[Test 10x10x10.zip](https://github.com/prusa3d/PrusaSlicer/files/13758444/Test.10x10x10.zip)

neophyl, 2023-12-23:
Yep thats output as a 0.01mm cube instead of a 0.01m cube.  That was opened in both Blender and also 3DTool.  Both open at the same size/scale.

sbussinger, 2023-12-26:
I submitted a bug to OnShape about this on 12/26. Too soon to have heard anything back yet. I'll leave a note here if I hear anything.

sbussinger, 2023-12-31:
I did a bit of digging into the STL file format and viewed the raw STL files generated by OnShape. The problem is that the STL file format doesn't even define a unit of measurement, it's just raw unitless coordinates. There's an 80 byte header which could have been used for this purpose but its contents are undefined and OnShape fills it with 0's.

So the short answer is that OnShape isn't doing it wrong, it's a poorly designed file format. My guess is that all the tools assume that the coordinates in the file are in millimeters and that's why if you export them from OnShape in millimeters everything works. If you export it from OnShape in Meters, the coordinates are all scaled by 1000 but the tools all still assume they are millimeters and get confused.

The short answer is export files from OnShape with the units set to Millimeter and pretend the other option doesn't exist.

Bilsid, 2024-05-20:
This is happening in Blender also, It's been happening since the last upgrade of prusa slicer, for about three months, except it has caused me a problem for the first time and won't work. I think this is a problem with Prusa slicer since I get the same thing from objects loaded from openscad

closed 2026-09-29 by prusaslicer-bot[bot]