prusa3d/PrusaSlicer #11929: no layers wer detected

F-CJPJ, 2023-12-17:
### Description of the bug

with versions 2.7.0 and 2.7.1; I tried to slice 2 parts and get the error "no layers detected".
using superslicer 2.4 or slic3r, no issue to slice.
the parts were designed and created using Onshape (trusted software).

### Project file & How to reproduce

[Rig_de_cintrage.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/13696452/Rig_de_cintrage.stl.zip)

 message is :
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

macos 14.1.2

### Printer model

wanaho D12

Jan-Soustruznik, 2023-12-18:
Hello, @F-CJPJ , 

could you share project(3MF) file where the problem is occur, please? Thank you

F-CJPJ, 2023-12-18:
Hello Jan

Finally I succeeded to export as 3MF files

[Link.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13706390/Link.3mf.zip)
[Rig_de_cintrage.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13706392/Rig_de_cintrage.3mf.zip)

closed 2026-09-29 by prusaslicer-bot[bot]