prusa3d/PrusaSlicer #11933: Need to print single inner and outer walls - but NO INFILL

huelli89, 2023-12-18:
### Description of the bug

PS don't let me print only an inner and outer wall. It adds support or infill structure where I don't want to have them!

### Project file & How to reproduce

PS don't let me print only an inner and outer wall. It adds support or infill structure where I don't want to have them!

I don't want to have these red lines. I said only 1 shell and 0 % infill.

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11

### Printer model

Anycubic Kobra 2 Max

neophyl, 2023-12-18:
Duplicate of several existing issues for example https://github.com/prusa3d/PrusaSlicer/issues/11246

Jan-Soustruznik, 2023-12-18:
Hello, Thank you for your report, but we know about this request. I Closed your ticked as duplicate. 

https://github.com/prusa3d/PrusaSlicer/issues/10102

thank you, have a nice day

closed 2023-12-18 by Jan-Soustruznik